renode/renode #368: [question] How to load the SampleCommandPlugin?

OliverMaster, 2022-08-11:
Hello, 
I have written a TestCommandPlugin based SampleCommandPlugin in Renode however, the command plugin are not being working as the following.
```
using System;
using Antmicro.Renode.UserInterface.Commands;
using Antmicro.Renode.UserInterface;
using AntShell.Commands;
using Antmicro.Renode.UserInterface.Tokenizer;

namespace Antmicro.Renode.Plugins.TestCommandPlugin
{
    public sealed class HelloCommand : Command
    {
        public override void PrintHelp(ICommandInteraction writer)
        {
            base.PrintHelp(writer);
            writer.WriteLine();
            writer.WriteLine("Usage:");
            writer.WriteLine(String.Format("{0} \"name\"", Name));
        }

        [Runnable]
        public void Run(ICommandInteraction writer, StringToken name)
        {
            writer.WriteLine(String.Format("Hello, {0}!", name.Value));
        }

        public HelloCommand(Monitor monitor) : base(monitor, "Test", "Greets a user.")
        {
        }
    }
}
```
```
Renode, version 1.12.0.37823 (44d6786a-202104022100)
(monitor) include @TestCommandPlugin.cs 
(monitor) Test
No such command or device: Test
(monitor) Test "user"
No such command or device: Test
(monitor) 
```
Any help will be highly appreciated.
Thanks in advance.

PiotrZierhoffer, 2022-08-12:
@OliverMaster there are two things to unpack here.

First of all, you're not really loading a plugin, you're compiling your code in runtime and want to use the command. This is a perfectly good approach, but there is no "initialization" code here. To have this command loaded automatically, change this line:

```
public sealed class HelloCommand : Command
```
to
```
public sealed class HelloCommand : AutoLoadCommand
```

Secondly, if you want to have a "proper" plugin, you need to have an "entrypoint" class with an attribute, like this: https://github.com/renode/renode-infrastructure/blob/master/src/Plugins/SampleCommandPlugin/SampleCommandPlugin.cs#L15

Then you can have the initialization code like this: https://github.com/renode/renode-infrastructure/blob/master/src/Plugins/SampleCommandPlugin/SampleCommandPlugin.cs#L20-L22

I would definitely go with the first approach though  - just include your .cs file.

OliverMaster, 2022-08-15:
Oh,Yes! I've tested the first approach, it works! 
I will close this issue. Thanks a lot.

OliverMaster, 2022-08-15:
@PiotrZierhoffer Oh,Yes! I've tested the first approach, it works!
I will close this issue. Thanks a lot.

closed 2022-08-15 by OliverMaster