renode/renode #370: Importing numpy (or other system modules)

mkschreder, 2022-08-17:
# Description

It would be very useful to be able to import system python modules such as numpy into peripherals. 

# Usage example

- Main use is to be able to do anything inside the extensions and not just things allowed by the limited IronPython environment

PiotrZierhoffer, 2022-09-28:
I have some initial success in that - I was able to import host modules by expanding the search path of the Python engine. I'm not yet sure how to reliably gather relevant paths though, but I'm looking into this

mkschreder, 2022-09-29:
@PiotrZierhoffer I decided to just go with the "external process" method and implemented a simplified version of socket based peripheral. Proof of concept here: https://github.com/swedishembedded/instruments and here is my simplified socket based peripheral plugin: https://github.com/swedishembedded/sdk/blob/main/renode/SocketPeripheral.cs

I like it so far. This way I'm not constrained by the limitations of renode itself and it's possible to link to any external libraries.

PiotrZierhoffer, 2022-09-29:
I believe you wanted to put that comment in https://github.com/renode/renode/issues/367 ?

DanieleParravicini-Synthara, 2024-02-22:
indeed it would be cool to be able to install and import other modules ;) 

anybody has any idea on how to do that?