renode/renode #371: cortex-a7 is  not supported?

chumen77, 2022-08-18:
cpu: CPU.Arm @ sysbus
    cpuType: "cortex-a7"

[err.log](https://github.com/renode/renode/files/9372566/err.log)

When I start, this error will be reported.

Inner exception:
Unknown cpu type

PiotrZierhoffer, 2022-10-18:
Hi @chumen77 .

While we do not explicitly support it, it should be compatible with `cortex-a9`. Please use it for now. I believe we might update this in future, but I can't give you ETA on this.

What we will do is to change the behavior not to throw the exception, but logging an error instead

PiotrZierhoffer, 2022-10-19:
@chumen77 FYI latest master will print an error in the Monitor rather than crash when you use `cortex-a7` type

chumen77, 2022-10-19:
Thank you very much for your reply, I will make more attempts.

PiotrZierhoffer, 2026-07-01:
Basic support for "cortex-a7" works right now. You can find a sample repl we use for the `opi_zero` target in Zephyr here: https://zephyr-dashboard.renode.io/zephyr_sim/769c4f2846db675bc5c584b6d35551ad239c72da/69ac352227821ca8b29c15396709038725064d8a/opi_zero/hello_world/hello_world.repl

closed 2026-07-01 by PiotrZierhoffer