renode/renode #372: Renode docker image having renode version `1.12`

John15321, 2022-08-19:
I was looking for an official Docker image for Renode and I can see that it hasn't been updated for over a year and has Renode `1.12`. Is this supposed to be like that for some reason or is there simply no effort planned to keep this image up to date?

https://hub.docker.com/r/antmicro/renode

PiotrZierhoffer, 2022-08-29:
It's an error, we'll work on pushing a new one soon. Thanks for pointing this out!

ok-mo, 2022-11-27:
@PiotrZierhoffer Hi Piotr, 

if it helps, I've been playing around with docker builds, especially larger builds that use caching for improved build performance. 

You can finds an example here:
https://github.com/ok-mo/se-sdk-develop/blob/main/.github/workflows/ci.yml

It is relatively straight-forward (when you know how I suppose!). It builds incrementally from the docker cache stored on docker hub with the image. 

You can find a recent build of a relatively large image here:
https://github.com/ok-mo/se-sdk-develop/actions/runs/3555631504/jobs/5972440327

and the image published to docker hub here:
https://hub.docker.com/r/okmo/develop/tags

You can probably get it to work with 3 docker actions
```
      - name: Set up Docker Buildx
        uses: docker/setup-buildx-action@v2
      
      - name: Login to Docker Hub
        uses: docker/login-action@v2
        with:
          username: ${{ secrets.DOCKERHUB_USERNAME }}
          password: ${{ secrets.DOCKERHUB_TOKEN }}

      - name: Build and push CI docker image
        uses: docker/build-push-action@v3
        with:
          context: .
          file: Dockerfile.build
          push: ${{ env.DOCKER_PUBLISH }}
          tags: okmo/build:latest
          cache-from: type=registry,ref=okmo/build:buildcache
          cache-to: type=registry,ref=okmo/build:buildcache,mode=max
```

I think it only publishes when the image has changed
(you can probably replace `${{ env.DOCKER_PUBLISH }}` with true