renode/renode #373: Unable to utilize Zephyr Network Stack on Renode

brent7984, 2022-08-22:
Hi, I am new to Renode, and I have noticed that **_renode/scripts/multi-node/quark-c1000-zephyr/demo.resc_** uses Zephyr's Echo_Server/Client Samples. 

It seems that Zephyr no longer supports the Quark board, so I attempted to build the Echo_Server/Client Samples using the NRF52840dk with the 802.15.4 overlay files. I tested the samples with actual dev kits and it worked (bilateral communication). 

However when attempting to use those elf's in Renode, I noticed that no packets was being sent to the nrf radios.

I have done the following:
 - Compiled Zephyr's Echo_Server/Client samples using "_**west build -p auto -b nrf52840dk_nrf52840 -- -DCONF_FILE="prj.conf overlay-802154.conf**_" 
 - Created wireless medium for 802.15.4 in renode and connected client and server's radio together
 - Logged 802.15.4 traffic using wireshark

My renode script is as follows:

```
emulation CreateIEEE802_15_4Medium "wireless"

######################### UDP SERVER - begin #########################
$bin=@/home/brenton/zephyrproject/renode_examples/renode_server/build/zephyr/zephyr.elf
$name="server"
i @scripts/single-node/nrf52840.resc
connector Connect radio wireless
mach clear
########################## UDP SERVER - end ##########################

######################### UDP CLIENT - begin #########################
 $bin=@/home/brenton/zephyrproject/renode_examples/renode_client/build/zephyr/zephyr.elf
 $name="client"
 i @scripts/single-node/nrf52840.resc
 connector Connect radio wireless
 mach clear
########################## UDP CLIENT - end ##########################

emulation LogIEEE802_15_4Traffic
```
renode_client/server is basically an exact copy of echo_client/server. 

I have noticed when I run the renode script, I get warnings about unhandled writes to nvic registers and failed to register PPI. From reading up on past comments I assume this is normal and not an issue.

I have also noticed the following by comparing the dev kits terminal to the renode terminal:

1. When typing "kernel stacks" in the renode terminal and the nrf dev kit terminal, I've noticed the following processes were missing in renode.
```
 udp6
 udp4_thread_id
 conn_mgr
```
2. The renode application does not make any network connections and it crashes when attempting to view certain network parameters within the terminal "net commands". The dev kit shows that a connection was established.

3.  typing "net iface" in renode shows the following:
```
      IPv6 unicast addresses (max 3):
      fe80::f6ce:3600:0:0 autoconf tentative infinite
```
                                                      **then terminal crashes/freezes**
    The dev kit shows the following:
```
IPv6 unicast addresses (max 3):
      fe80::f6ce:368a:59b1:6b43 autoconf preferred infinite
      2001:db8::1 manual preferred infinite
IPv6 multicast addresses (max 4):
      ff02::1
      ff02::1:ffb1:6b43
      ff02::1:ff00:1
```
4.  On renode, typing "net ipv6", shows the following then it crashes:
```
    IPv6 addresses for interface 1 (0x20002d80) (IEEE 802.15.4)
    Type       State     Lifetime (sec) Address
```
    The dev kit shows:
```
    IPv6 addresses for interface 1 (0x20002d80) (IEEE 802.15.4)
    Type        State       Lifetime (sec)    Address
    autoconf    preferred   infinite          fe80::f6ce:368a:59b1:6b43/128
    manual      preferred   infinite          2001:db8::1/128
```

I am a bit confused as to why the renode application does not detect a network interface. I assumed my approach would work as I followed the guidelines on "setting up a wireless network" on Renode and I also looked into the Quark sample which used the Zephyr Echo_Client/Server samples, and that script didn't do any config relating to networking.

I assume my issue should be solved with a simple config, which I cannot seem to find in the Renode documentation. Could someone please inform me on how to set up the network infrastructure for my application.

Renode uses the following elf's for its provided samples.
**$bin=@https://dl.antmicro.com/projects/renode/zephyr-echo-server.elf-s_1003060-56eb87aafc8c00f73ebe154dafee627a3a4f23f6
$bin=@https://dl.antmicro.com/projects/renode/zephyr-echo-client.elf-s_1016316-bfb6584d9d63950e8106c6820efccdaf0b7738a5**

Is it possible for people to view the source code of these provided samples so that we can analyze what changes were made to get them working with Renode?

With thanks
Brenton

mateusz-holenko, 2022-09-27:
This is related to #363.

brent7984, 2022-09-27:
I've noticed that Renode gives warnings about unimplemented registers:
- NVMC:ICACHECNF
- EGU3:INTENSET
- EGU3:INTENCLR
- TEMP:INTENSET
- TEMP:TASKS_START

Does anyone know if these registers are required for 802.15.4 communications.
I've tested the BLE stack on Zephyr, and that works fine. For BLE samples, the following registers were also listed as unimplemented in Renode:

- NVMC:ICACHECNF
- TEMP:INTENSET