renode/renode #374: VexRISCV-SMP script has inbuilt rootfs.cpio

amr-25, 2022-09-01:
Hi,

I want to run the executable in renode that is simulating vexriscv+smp+linux. In board I usually generate the rootfs.cpio through buildroot that has the executable. But the script provided with renode has a link which downloads the rootfs.cpio (link is https://dl.antmicro.com/projects/renode/litex_vexriscv_smp--rootfs.cpio-s_4570112-7a6a6388e09170db38795a006dd75f91d556eecf). If  I manually download, add files to the archive, and add it in the script, simulation stop at step initialiasing sram. Is there any way to rectify this? or Should I use litex2renode generator, generate platform repl and edit the resc script to include linux+rootfs.cpio etc. ?

Thank you

MateuszKarlic, 2022-09-15:
Hi @amr-25 ,

I tried to reproduce your issue, and I could extend the rootfs with no problems.
Hovewer, please note, that the maximum size of rootfs is restricted by a parameter in the Device Tree. For this instance it is defined as:
```
        chosen {
                linux,initrd-start = <0x41000000>;
                linux,initrd-end = <0x41800000>;
        };
```
so if your rootfs is bigger than that, Linux might crash. You need to decompile the dtb, edit this parameter and recompile it.

Another possibility is that your rootfs has invalid format. I used the following commands to create it:
```
mkdir rootfs && cd rootfs
cpio -idv < ../litex_vexriscv_smp--rootfs.cpio

# add new files here

find . -print0 | cpio -ov --null --format=newc > ../litex_vexriscv_smp--rootfs.cpio
```
You should remember to run these as a root user, or issues can appear with the `/dev` subsystem when packaging/unpackaging.

amr-25, 2022-09-19:
Thank you for the response. I got the rootfs with files in renode using the commands you shared. I added binaries(a simple hello world) and it boots fine. But when I run the binaries(a simple hello world), I get Glibc version 2.34 not found error. I think it is problem with the version of the linux buildroot. Are you aware of the buildroot config to generate the binaries?

MateuszKarlic, 2022-09-22:
Binaries used in the demo shipped with Renode were built some time ago, so they target an older kernel. We plan to update the demo, but for now I would suggest building all components from scratch as described in `linux-on-litex-vexriscv` repository's Readme.

Note: we noticed problems when building against the buildroot master, but were able to successfully build it using the `2022.08` tag. That will give you a recent kernel, device tree blob, toolchain, and rootfs, that you could extend using the same commands as above.

In order to run the whole setup you'll also need a newer `opensbi.bin`. It can be built from https://github.com/litex-hub/opensbi, branch `0.8-linux-on-litex-vexriscv`, but I am attaching it here for convenience as well.

We are currently working on updating the `litex_json2renode` script so that it can generate a platform definition for the SMP configuration automatically, but for now I attach script files compatible with the 4 cores setup.

[vexriscv-smp.zip](https://github.com/renode/renode/files/9624928/vexriscv-smp.zip)