renode/renode #377: emulation LogIEEE802_15_4Traffic failed to detect Wireshark path

ekawahyu, 2022-09-07:
# Description

Executing `emulation LogIEEE802_15_4Traffic` gives an error as follows:

> There was an error executing command 'emulation LogIEEE802_15_4Traffic'
> Wireshark is not installed or is not available in the default path. Please adjust the path in Renode configuration.

The path to Wireshark has been added by executing:

> path add @/Applications/Wireshark.app/Contents/MacOS

Here is the PATH:

> Current 'PATH' value is: /Applications/Renode.app/Contents/MacOS;/private/tmp/contiki-ng/examples/rpl-udp;/Applications/Wireshark.app/Contents/MacOS

# Expected behaviour

Wireshark should show up for traffic analysis.

# How to reproduce?

I use Renode with Contiki-NG, here is the simple steps to reproduce it:

- git clone https://github.com/contiki-ng/contiki-ng
- git submodule update --init --recursive
- cd examples/rpl-udp
- make renode TARGET=cc2538dk

# Environment

macOS Monterey 12.4 on MacBook Pro M1
gcc-arm-embedded 10.3.1 20210824 (Homebrew)
wireshark version 3.6.7 (v3.6.7-0-g4a304d7ec222) (Homebrew)

# Do you plan to address this issue and file a PR?

No, because I haven't been able to debug this issue. How can I turn on debug messages while executing Renode?

PiotrZierhoffer, 2022-09-13:
Hi @ekawahyu , thanks for the report.

 The error message is not clear enough.

Renode config is located in ~/.config/renode/config on macOS.

You will find the `wireshark-path` entry there. Try to adjust it to the path you need and please let me know if it helps.

ekawahyu, 2022-09-13:
@PiotrZierhoffer, thanks for pointing out where to manually modify the `wireshark-path`. That helps, but I am wondering if there is a way to override the default path through command line instead of manually change it in the config file. Thank you and you can close this issue.

ekawahyu, 2022-09-13:
On a second thought, if the `Wireshark-path` is updated from `/usr/bin/wireshark' to '/usr/local/bin/wireshark`, any macOS user can create symbolic link to Wireshark.app without bypassing the System Integrity Protection (SIP) on macOS. I just created one and it seems to work as expected.

PiotrZierhoffer, 2022-10-11:
What do you think about setting the macOS default to just `/Applications/Wireshark.app/Contents/MacOS/Wireshark`? It works nicely, but I'm not sure how generic would that be.

ekawahyu, 2022-10-13:
If the config file is specific to macOS, then yes, that should be generic enough. But if it was intended to be generic across multiple platforms, then `/usr/local/bin/wireshark` is best. That's my opinion.

PiotrZierhoffer, 2022-10-13:
@ekawahyu this is MacOS specific. We have changed it in code, but to see the effect for yourself you will need to remove the config file and let it be regenerated.

We have also made the error message more precise.

These changes will be available in the next nightly package at https://builds.renode.io .

closed 2022-10-13 by PiotrZierhoffer