renode/renode #378: Endianness translator

andresdemski, 2022-09-08:
I'm trying to simulate a system which has a big-endian cpu and several little-endian and big-endian peripherals. I got it working modifying the C# models and assigning them bigendianness.

In the hardware we are using, the endianness translation is done in the bus crossbar/interconnect. Is there any way to model this behavior without modifying any of the peripherals already supported by renode?

Thanks you

mateusz-holenko, 2022-09-13:
Hi @andresdemski, that's an interesting topic.

At the moment automatic endianness translation can be achieved by marking peripheral models with [the atrribute](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Main/Peripherals/EndianessAttribute.cs) (which I assume it's what you've already tried).

We are currently looking into extending the bus implementation to support dynamic setups, though. We'll let you know once we have a working solution.

andresdemski, 2022-09-14:
Let me know if i can help you. I'm not good coding c# but maybe I can help with testing.

Thank you for this amazing tool!

andresdemski, 2022-11-22:
Hey, me again. 

Is there any way to override that attribute or change endianess in runtime? I want to modify it at least from the REPL instantiation.