renode/renode #379: icicle-kit example wont boot; hangs when trying to start kernel

Mai-Lapyst, 2022-09-11:
# Description

When starting the `icicle-kit.resc` script, the emulation runs until it prints out `Starting kernel ...` on mmuart1 and then stops. The log indicate that the CPU's u54_1, u54_3 and u54_4 are aborted due to an error: `Trying to execute code outside RAM or ROM at 0xDB078793FFFD9795.`

# Expected behaviour

That the icicle-kit example boots up correctly linux. Instead it hangs when 3 simulated CPU's stop.
Last few lines of log:
```
23:39:18.3650 [WARNING] u54_1: Trying to set unaligned STVEC: 0xFFFD9797, aligning to 4-byte boundary.
23:39:18.3650 [WARNING] u54_1: STVEC value written to CSR corrected from 0xfffd9797 to 0xfffd9795 because of the selected interrupt mode and privilege architecture
23:39:18.3657 [WARNING] sysbus: [u54_1: 0xDB078793FFFD9795] ReadByte from non existing peripheral at 0xDB078793FFFD9795.
23:39:18.3659 [ERROR] u54_1: CPU abort [PC=0xDB078793FFFD9795]: Trying to execute code outside RAM or ROM at 0xDB078793FFFD9795.
23:39:18.3671 [WARNING] u54_3: Trying to set unaligned STVEC: 0xFFFD9797, aligning to 4-byte boundary.
23:39:18.3671 [WARNING] u54_3: STVEC value written to CSR corrected from 0xfffd9797 to 0xfffd9795 because of the selected interrupt mode and privilege architecture
23:39:18.3677 [WARNING] sysbus: [u54_3: 0xDB078793FFFD9795] ReadByte from non existing peripheral at 0xDB078793FFFD9795.
23:39:18.3678 [ERROR] u54_3: CPU abort [PC=0xDB078793FFFD9795]: Trying to execute code outside RAM or ROM at 0xDB078793FFFD9795.
23:39:18.3725 [WARNING] u54_4: Trying to set unaligned STVEC: 0xFFFD9797, aligning to 4-byte boundary.
23:39:18.3725 [WARNING] u54_4: STVEC value written to CSR corrected from 0xfffd9797 to 0xfffd9795 because of the selected interrupt mode and privilege architecture
23:39:18.3732 [WARNING] sysbus: [u54_4: 0xDB078793FFFD9795] ReadByte from non existing peripheral at 0xDB078793FFFD9795.
23:39:18.3732 [ERROR] u54_4: CPU abort [PC=0xDB078793FFFD9795]: Trying to execute code outside RAM or ROM at 0xDB078793FFFD9795.
```
Full log (including output of mmuart0 & mmuart1): https://gist.github.com/Mai-Lapyst/99e060c96cb0ed34ad35863995a6771f

# How to reproduce?

- Install renode
- Run `renode /opt/renode/scripts/single-node/icicle-kit.resc`
- Type `s` or `start` in the monitor
- Wait until HSS and U-Boot are done, and it tries in mmuart1 to load the kernel; i.e. "Starting kernel ..." is visible on the screen.

Files used:
- [icicle-kit.resc](https://github.com/renode/renode/blob/a133ada059deb6db9f32f795319949f7fcaae133/scripts/single-node/icicle-kit.resc)
  - [polarfire-soc.resc](https://github.com/renode/renode/blob/a133ada059deb6db9f32f795319949f7fcaae133/scripts/single-node/polarfire-soc.resc)
    - [polarfire-soc.repl](https://github.com/renode/renode/blob/a133ada059deb6db9f32f795319949f7fcaae133/platforms/cpus/polarfire-soc.repl)

# Environment

Please, provide the following information:

- OS: Arch linux; 5.19.7-arch1-1
- Renode version: 1.13.1.34293 (a133ada0-202207251903)

Installed via AUR:  https://aur.archlinux.org/packages/renode-bin

# Additional information

None

# Do you plan to address this issue and file a PR?

No

PiotrZierhoffer, 2022-09-27:
Hi @Mai-Lapyst 

We can confirm this behavior and have found out the reason why it didn't fail our tests.

As we are working on a solution, there is a workaround you might try. It's, quite manual, but you can automate with e.g. hooks

When you reach the U-Boot console, run the following in the Monitor:

```
p
u54_4 ClearTranslationCache 
u54_3 ClearTranslationCache 
u54_2 ClearTranslationCache 
u54_1 ClearTranslationCache 
e51 ClearTranslationCache 
s
```

Mai-Lapyst, 2022-09-27:
Hi, thanks for the response. The workaround works for me, thanks again!

Shivaraj199, 2022-10-06:
> Hi @Mai-Lapyst
>
> We can confirm this behavior and have found out the reason why it didn't fail our tests.
>
> As we are working on a solution, there is a workaround you might try. It's, quite manual, [...]

Hello @PiotrZierhoffer
I am also facing the same issue when i Reach the Uboot console I cant able to enter the folloewing commnads it goes to 

RISCV#
RISCV#
RISCV#
How can i enter the following commands that you specified above

thank you

Micko8, 2022-10-06:
> > Hi @Mai-Lapyst
> > We can confirm this behavior and have found out the reason why it didn't fail our tests.
> > As we are working on a solution, there is a workaround you might try. It's, quite manual, but you can automate with e.g. hooks
> > When you reach the U-Boot console, run the following in the Monitor:
> > ```
> > p
> > u54_4 ClearTranslationCache 
> > u54_3 ClearTranslationCache 
> > u54_2 ClearTranslationCache 
> > u54_1 ClearTranslationCache 
> > e51 ClearTranslationCache 
> > s
> > ```
> 
> Hello @PiotrZierhoffer I am also facing the same issue when i Reach the Uboot console I cant able to enter the folloewing commnads it goes to @Micko8 RISCV# RISCV# RISCV# How can i enter the following commands that you specified above
> 
> thank you

@Shivaraj199 
This command is a Renode one. 
That worked:

In uuart1 monitor:

Hit any key to stop autoboot:  0 
RISC-V # 

Renode Monitor:

(machine-0) p
Pausing emulation...
(machine-0) u54_4 ClearTranslationCache
(machine-0) u54_3 ClearTranslationCache
(machine-0) u54_2 ClearTranslationCache
(machine-0) u54_1 ClearTranslationCache
(machine-0) e51 ClearTranslationCache
(machine-0) s

In uuart1 monitor:

RISC-V # boot