renode/renode #380: Add support for STM32H725/735 cpu

escherstair, 2022-09-13:
I've just read about renode project and I've been trying to learn it.
I see a lot of CPU in https://github.com/renode/renode/tree/master/platforms/cpus
Is it possible adding support for [STM32H725/735](https://www.st.com/en/microcontrollers-microprocessors/stm32h725-735.html) cpu?
ST has a [Discovery Kit](https://www.st.com/en/evaluation-tools/stm32h735g-dk.html) too.

Thanks

PiotrZierhoffer, 2022-09-13:
Hi @escherstair 

H7 series is not in our immediate scope at the moment. As ST platforms are generally similar (have similar peripherals, it should not be very difficult to add - we're happy to accept contributions and we can provide a few guidelines where to start.

If you have commercial interest in supporting this platform in Renode, please reach out to contact@renode.io.

akohlsmith, 2023-07-11:
I have been poking around at adding support for various bits and bobs on the H735. I wrote a stupid python RCC which is enough to allow the CubeMX HAL to think it's running on a real machine. My changes are just in the `.repl` so far, but adding support for MPU and some of the hairier peripherals will require something more than a python script.

PiotrZierhoffer, 2023-07-31:
@escherstair it's been some time, but I'm happy to report that we have support for STM32H743.

You can also see on the Zephyr Dashboard (https://zephyr-dashboard.renode.io/) that H735 is supported as an autogenerated platform, to the extent that it runs several Zephyr binaries successfully. You can search for "ST STM32H735G Discovery". You can also find the Discovery board as supported in Renodepedia (https://zephyr-dashboard.renode.io/renodepedia/boards/stm32h735g_disco/).

I will close this issue, if you need to support something specific - feel free to open a new one.

@akohlsmith please take a look at https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Miscellaneous/STM32H7_RCC.cs

As with most RCC peripherals, this is a stub, but you should use this to move forward.
In general I'd suggest using one of our nightly builds (https://builds.renode.io) or build from sources for the latest features (including MPU support on Cortex-M tested with Zephyr).

closed 2023-07-31 by PiotrZierhoffer

akohlsmith, 2023-08-24:
@PiotrZierhoffer I appreciate the fact that you've got a very slick one command download and run demo, but it's hiding all of the details and even as slick as it is, it fails to run on osx, with renode-run complaining that it can't download renode automatically. I don't need it to download anything automatically, I have renode installed, but poking around renode-run doesn't give me any hints on how to bypass the attempt to automatically download and use the already-installed version. Further, the renodepedia doesn't give me a link to the code/project that renode-run is going to automatically download and run for me, meaning that unless the user is doing things in exactly the same way that the point-and-click demo is expecting, there isn't any way to even look around.

akohlsmith, 2023-08-24:
Aha, Not being a Zephyr person I had assumed the artifacts were what you collected AFTER you ran the test (logs, binaries, etc.). Thank you.