renode/renode #381: I2C Connector

andresdemski, 2022-09-14:
Hi! First of all, RENODE IS AMAZING!! Thank you for open this amazing tool.

# Problem

I'm trying to connect to machines using a I2C bus. 

# Question
Is there any way to connect two I2C peripherals from different machines? or should I develop a custom implementation for that?
What do you recommend?

Thank you.

PiotrZierhoffer, 2023-04-19:
Hi @ydnatag. Thanks for the kind words! Sorry that it took us a moment to reply. After some thought what you asked about seems possible, with some restrictions.

Usually, connections between machines in Renode are delayed to guarantee determinism. This way, no matter how occupied your host is, you can have the same results in tests.

I2C in Renode is conceptually synchronous, as we’re working on a fairly high level of abstraction.

There are three ways this could be addressed:

* A proxy that will buffer device data before sending it to the bus controller - this is relatively straightforward to implement, but will only work for a specific model of communication
* A “full” two way communication - this will violate determinism guarantees, but for many cases it’s not a big issue
* Revamp of the I2C interface. While conceptually more difficult, it might benefit many similar use-cases.

Can you share some more details about your use case?

If you prefer not doing it publicly, happy to switch to e-mail.

ivklink, 2025-01-06:
Hi @PiotrZierhoffer 

We are looking at using Renode to emulate system management using IPMB (I2C). Hence we are looking for I2C connector between two machines, would be great if you can advise us on the feasibility of this. It might be a powerfull feature for emulating system management for backplane based embedded systems.

Happy to share more details about our use case too.

Thank you

ivklink, 2025-02-13:
Hi @PiotrZierhoffer 

Just want to check if you have received my message above? It would be great if an I2C connector can be made, perhaps with some guidance we can contribute here.

Thank you