renode/renode #382: Documentation for creating multi node boards

sjalloq, 2022-09-17:
### Description

I'm wanting to understand the flow for creating multi node boards but couldn't see anything in the official documentation. There's also no description of how interfaces are modelled.

My intent is to create a system including one of the STM32 boards with a custom Verilated MCU connected via a bidirectional 2 wire interface.
 
Are there any examples or tutorials of multi node setups?

Thanks.

sjalloq, 2022-09-29:
So I think I've answered some of this myself now after finding similar questions.  Issue #296 has a very similar issue to solve so I'll try that for the GPIO connections.  In terms of connecting an SPI master to an SPI slave, I can't see any examples or work out what the correct method is.  There doesn't seem to be an SPI connector to use.  What am I missing?

mgielda, 2022-10-17:
Hi @sjalloq - interesting use case! Yes, we're updating the docs with more info around those topics. There is a new chapter here which you might find interesting: https://renode.readthedocs.io/en/latest/networking/machine-to-machine-connections.html

If you see more info that would be useful to add to this chapter, perhaps you could help with a PR to the docs? (it's the renode-docs repo)