renode/renode #384: Help for adding new cpu architectures

xsession, 2022-09-23:
Hello!

I've just read about the renode project it's really impressive!
I want to start my embedded systems with renode, but i use a bunch of a bit exotic cpu cores (microchip dsPIC33, dsPIC30, AVR, texas instruments C28). I try to look ahead how can i implement its, but in the case of peripherial its clear the concept, but for me, in the CPU case is not.
Can you help me some starting point for it with some simple example/template/docs?

PiotrZierhoffer, 2022-09-27:
Hi @xsession

We do not have specific documentation covering this topic, but please take a look at https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/CPU/ExternalCPU.cs

This is a class you can copy and fill out the places with comments with your logic - it should be enough to perform basic execution from a given CPU.

Of course the implementation itself is a different thing - there are many possible approaches and you have to choose the right one for you. 

Our implementations base either on the tlib submodule (https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/CPU/TranslationCPU.cs) or on verilated CPUs (https://github.com/renode/renode/blob/master/src/Plugins/VerilatorPlugin/Verilated/Peripherals/VerilatedCPU.cs)

If you plan to implement the logic of the CPU in C, then you might like the https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Main/Utilities/Binding/NativeBinder.cs tool - you will see how we use it in both our CPU implementations, it helps calling C from C# and the other way around.

septatrix, 2022-09-27:
Would it be possible to somehow integrate this with avrsim using the C wrapper you mentioned?

xsession, 2022-09-29:
Hi @PiotrZierhoffer

Thank you for your respond, i will try to implement a core based on the examples. First maybe the 8-bit avr, i find a low level implementation example in qemu and then try to bind.

PiotrZierhoffer, 2022-10-11:
> Would it be possible to somehow integrate this with avrsim using the C wrapper you mentioned?

In principle, probably yes, but we have never worked with it. We are not planning to include it as part of Renode, as it's license is incompatible with our model. But technically I don't see reasons why it shouldn't work.

@xsession good luck! I'll close this issue, please feel free to open a new one if needed.

closed 2022-10-11 by PiotrZierhoffer