renode/renode #385: Peripheral modelling questions - base class choices

shareefj, 2022-09-28:
I'm trying to model a peripheral and have a few questions which I'll try to create separate tickets for to avoid polution.

1.  What are the use cases for `SimpleContainer` and `NullRegistrationPointPeripheralContainer` and when should one be used over the other?  I was looking at the SPI peripherals as an example and can't immediately work out the differences.  There aren't any comments in the code and there's no documentation describing the base classes.  Is there a plan to fix this or am I missing some tome of information somewhere?

PiotrZierhoffer, 2022-09-29:
The simple answer is:

- `SimpleContainer` accepts many peripherals as children, with their respective numbers. This is a common case in I2C.
- `NRPPContainer` takes only one peripheral. This we often use for SPI controllers, unless we have some specific chip select scenarios we want to support (or the controller itself controls the CS line)

For `SimpleContainer` the registration looks like this (`1` is the number of the child):

```
peripheral: Type @ container 1
```

For `NullRegistrationPoint...`:
```
peripheral: Type @ container
```

shareefj, 2022-09-29:
Thanks for that.  So the use case for these containers is when you're connect them within a larger system, i.e. `SimpleContainer` would allow you to connect multiple I2C sensors to a single I2C master?

On a related note, I've been trying to understand how I go about defining a peripheral outside of the main `renode-infrastructure` source tree.  I randomly found an example of someone using an include statement in their resc file, `i @my_local_dir/peripherals/my_new_peripheral.cs`.  That would be a useful point to document.  But given I'd copied the examples of wrapping `ISPIPeripheral` in `NRPPContainer`, I found that when I tried to include it I got an error that nothing was known about `machine`.  The examples I found didn't seem to bother wrapping `ISPIPeripheral` and instead used `var machine = this.GetMachine()` when needed.  Is there any guidance you can give for this use case of loading local peripherals.

PiotrZierhoffer, 2022-10-11:
> Thanks for that. So the use case for these containers is when you're connect them within a larger system, i.e. `SimpleContainer` would allow you to connect multiple I2C sensors to a single I2C master?

This is correct

> On a related note, I've been trying to understand how I go about defining a peripheral outside of the main `renode-infrastructure` source tree. I randomly found an example of someone using an [...]
I don't fully understand what's the issue without seeing the code or the error itself.
Generally you can `include @your/peripherals.cs` if we already know about all other types they use.
We also analyze a certain subset of loaded files - for example peripherals.

I assume you're implementing an I2C controller. Using `NRPPContainer` alone is not enough. You need to also add e.g. `IDoubleWordPeripheral` interface to allow it to connect to system bus, if I understand your use case correctly.

> The examples I found didn't seem to bother wrapping `ISPIPeripheral` and instead used `var machine = this.GetMachine()` when needed. Is there any guidance you can give for this use case of loading local peripherals.

These are two different scenarios. `ISPIPerpipheral` will be used for a peripheral, and a `*Container` will be used for a controller.

shareefj, 2022-10-12:
Hi,

I was actually trying to implement an SPI peripheral but I guess there's not much difference.  I ended up with the following:

```
namespace Antmicro.Renode.Peripherals.SPI
{
    public class SimpleSPIPeripheral : ISPIPeripheral, IDoubleWordPeripheral, IKnownSize
    {
```

and then internally when I wanted to update the interrupt flags, called:

```
this.GetMachine().LocalTimeSource.ExecuteInNearestSyncedState(_ => UpdateInterrupts());
```

Is that sensible?  I think I now understand the use case of the different base classes and have moved on to trying to work out how time is consumed.

PiotrZierhoffer, 2022-10-18:
For SPI peripherals you don't need to worry about `*Container` classes at all - they are used for controllers only.

Regarding machine - an easier way to obtain the machine object (and in general a bit more robust) is to pass it as a constructor parameter.

So if you have:

```
public SimpleSPIPeripheral()
{
    // your constructor logic, if any is needed
}
```
you can change it to

```
public SimpleSPIPeripheral(Machine machine)
{
    this.machine = machine;
    // your constructor logic, if any is needed
}

private readonly Machine machine;
```

This parameter will be automatically populated when created the `SimpleSPIPeripheral` from the repl file, no need to change anything there.

And yes, you can use `ExecuteInNearestSyncState` like you do.

shareefj, 2022-11-07:
Thanks for this last update. I'll use that in the future.

closed 2022-11-07 by shareefj