renode/renode #387: Correct way of passing local repl files to robot tests

shareefj, 2022-09-29:
Hi,

when running Renode directly, I can use `@platforms/cpus/cpu.repl` as the path for the platform file I want to use.  However, when running a test using the Robot framework I can no longer use the `@` syntax because the directory changes to where the test sits.

So if I have the following directory structure:

```
platforms/cpu/cpu.repl
platforms/boards/mycpuboard.repl
tests/test1.robot
```

in `test1.robot` I seem to need to specify the platform file using:

```
*** Test Cases ***
My Fantastic Test
    Execute Command             mach create
    Execute Command             machine LoadPlatformDescription @${CURDIR}/../platforms/cpus/cpu.repl
...
```

Is there any way around this or is that just a byproduct of using Robot?

shareefj, 2022-09-29:
And what happens when you include `.resc` scripts that use `@` syntax?  They'll all fail too.

shareefj, 2022-10-12:
Hi @mateusz-holenko,

thanks for getting back to me.  Yes, the `platforms/boards` path I was referring to was my own.  Probably not the best naming but I was trying to replicate the Renode source tree in my local area.

So the use case would be to store additional user defined platform files locally within my repo and be able to run Robot tests against them.

PiotrZierhoffer, 2022-10-18:
If you're referring to files relative to your resc, you should use the `$ORIGIN` variable.

So your `system.resc` would have:

```
machine LoadPlatformDescription $ORIGINplatforms/cpus/cpu.repl
```

(please notice lack of `@` - this is related to the fact that Robot variables are not really controlled by us, they are provided by the tool, hence the inconsistency)

shareefj, 2022-11-07:
OK, thanks.  I've kind of lost track of this issue as I was experimenting in a temporary directory.  I'll assume I can use `$ORIGIN` and open a new ticket for any further issues.  Thanks.

closed 2022-11-07 by shareefj