renode/renode #389: Renode hangs if printing to UART analyzer too quickly

shareefj, 2022-10-02:
### Description

I've been trying to debug a number of issues with getting Renode working and can induce a hang when printing to a UART.  I'm guessing that you have some buffer overflowing when the sink can't empty the character buffer quickly enough.  If I run the same program with or without using `showAnalyzer usart` then I get a hang.  If I use `CreateUartPtyTerminal` and use screen to connect, then all is well.

### How to reproduce?

Run the template here: https://github.com/shareefj/uart-hang

### Environment

- OS : Ubuntu 20.04.3 LTS (WSL)
- Renode: Renode, version 1.13.1.34293 (a133ada0-202207251903)

PiotrZierhoffer, 2022-10-11:
Hi @shareefj .

Thanks for the report and a reproduction case. This is a known issue. We have implemented some improvements in this area recently, that help with many real life use cases, but the overall issue is still there and is not trivial to fully resolve.

We'll update this ticket when we have more progress in this area.

shareefj, 2022-11-07:
@mgielda - had no idea there was any build flow running so I haven't updated anything there.  The "missing" `repl` file is there in the repo.

closed 2022-11-07 by shareefj