renode/renode #390: GPIOConnector incorrectly calls sourcePin.Disconnect

shareefj, 2022-10-02:
### Description

I've been debugging a warning spat out when using the `GPIOConnector` class to connect GPIO between machines.  The warning is that on line [72](https://github.com/renode/renode/blob/master/src/Renode/Connectors/GPIOConnector.cs#L72).  

The issue is that this code block and that starting on line [65](https://github.com/renode/renode/blob/master/src/Renode/Connectors/GPIOConnector.cs#L65) is not needed.  The `GPIO` being selected as the `sourcePin` can have multiple connections so this method should just connect itself to the `sourcePin` and exit.  What's worse, the `sourcePin.Disconnect` method is called which disconnects any previously connected GPIO to this source.  In my case, using  the `stm32f4.repl` platform, the connection between the GPIO and the external interrupt module was dropped.

### Expected behaviour

The `GPIOConnector.SelectSourcePin` method should call `sourcePin.Connect` only and not call `sourcePin.Disconnect`.

### How to reproduce?

Use the GPIOConnector to connect GPIO between multiple machines.

```
using sysbus

emulation CreateGPIOConnector "gpio-con"

mach create
machine LoadPlatformDescription @platforms/cpus/stm32f4.repl
connector Connect gpioPortA gpio-con
gpio-con SelectSourcePin gpioPortA 0
mach clear

mach create
machine LoadPlatformDescription @platforms/cpus/stm32f4.repl
connector Connect gpioPortA gpio-con
gpio-con SelectDestinationPin gpioPortA 0
mach clear
```

### Do you plan to address this issue and file a PR?

PR incoming.

PiotrZierhoffer, 2022-10-11:
Hi @shareefj . I added some comments to #391

shareefj, 2022-11-07:
@PiotrZierhoffer We never resolved that conversation on the PR but I'm happy to close it as the suggested fixes were broken anyway.  However, can we bottom out the main issues please.

Just to recap:

1. Within `GPIOConnector.cs`, any call to `GPIO::Disconnect()` should also pass in the `GPIOEndpoint`.  If this doesn't happen, then every connected GPIO will be disconnected when `targets.Clear()` is called: https://github.com/renode/renode-infrastructure/blob/fb96e4c4c793bbf918aa262d60b91ecd4dddb855/src/Emulator/Main/Core/GPIO.cs#L67
2. There is a bogus warning emitted from https://github.com/renode/renode/blob/master/src/Renode/Connectors/GPIOConnector.cs#L70 when **any** GPIO is connected to the source pin.