renode/renode #393: Time source "Performance" property

Joshua-Shaffer, 2022-10-17:
According to Renode documentation at https://renode.readthedocs.io/en/latest/advanced/time_framework.html, time sources are supposed to contain a property called "Performance" which dictates the ratio of virtual time to real time. No methods appear visible in the monitor to modify this property. Looking through `TimeSourceBase.cs` in `/Infrastructure/src/Emulator/Main/Time` shows a few comments referencing a "Performance" property without the parameter existing in the object. Furthermore, the description for the `InnerExecute` method states 
`/// (4) (optional) sleep if the virtual time passed faster than a real one; this step is executed if <see cref="AdvanceImmediately"> is not set and <see cref="Performance"> is low enough`, 
but it does not appear that this feature exists in that method.

Given this information, I wanted to ask if the performance feature actually exists currently, or if it's a deprecated or planned feature.

Joshua-Shaffer, 2022-10-19:
Is the team aware of any caveats to the behavior of turning `AdvanceImmediately` on or off? Using the setup at https://github.com/antmicro/renode-rtems-leon3, I created a recipe to run a custom RTEMS binary and am seeing the load sit consistently below 1, despite setting `emulation SetGlobalAdvanceImmediately` to true or false. This results in virtual time always accumulating faster than real time. The default shell example appears to work correctly, though (although in `AdvanceImmediately off` config, the load will always sit just slightly above 1, and virtual time will slip slowly).

PiotrZierhoffer, 2022-11-15:
Yes, we are aware that with `AdvanceImmediately` we slip a little bit behind. This is something we are planning to address, but it might take a while.

Regarding the RTEMS demo - it seems to just not sleep, really. It's polling on registers instead, and this is a difficult operation from the Renode perspective.

PiotrZierhoffer, 2023-09-21:
The `Performance` property was removed from the docs quite some time ago, closing

closed 2023-09-21 by PiotrZierhoffer