renode/renode #394: Correct way to set flash size on NRF52

mkschreder, 2022-10-17:
### Description

There is an FICR register on NRF that contains factory information. (https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fficr.html) what is the correct way to set values in these registers for a particular simulation?

PiotrZierhoffer, 2022-10-18:
@mkschreder you will see that our current REPL for nRF52840 does not implement the FICR region - we only have a tag there (https://github.com/renode/renode/blob/master/platforms/cpus/nrf52840.repl#L98):

```
Tag <0x10000000, 0x10000FFF> "FICR"
```

To easily patch it you can use the `Tag` command to create a "subtag" with a proper return value:

```
sysbus Tag <0x10000110, 0x10000113> "INFO.FLASH" 0x80

mkschreder, 2022-10-19:
Ok. I solved it by adding arraymemory and then writing that region with correct values in the custom repl file.

closed 2023-01-10 by PiotrZierhoffer