renode/renode #395: How to set Chip Select for external SPI flash?

maximevince, 2022-10-19:
### Description

I'd like to be able to configure the Chip Select (CS) pin for an SPI peripheral.
In this case, it's a `Micron_MT25Q` SPI flash device.

There's other peripherals on the same SPI bus, so I suspect the `Micron_MT25Q` will try to interpret all data on that bus, if we don't setup the CS line in the simulator.

### Usage example

This is the relevant part of the `resc` file:
```
flash0: Memory.MappedMemory @ sysbus 0x0
    size: 0x80000

flash: SPI.Micron_MT25Q @ spi1
    underlyingMemory: flash_mem
    // how to configure chip select? -> gpio@6

flash_mem: Memory.MappedMemory
    size: 0x1000000 // 16 MBit
```

PiotrZierhoffer, 2022-10-24:
Hi @maximevince 

At this moment we don't have examples with CS on SPI. Our SPI implementation is point-to-point, but it can be addressed on the controller level.

If you'd want to modify the `NRF52840_SPI` controller, then what you'd need to do is:

- change `NullRegistrationPointPeripheralContainer` to `SimpleContainer`, allowing you to register flash by number (e.g. `flash: SPI.Micron_MT25Q @ spi1 0`). This number will be kinda arbitrary, it's just important to have separate numbers for all connected devices.
- add `private ISPIPeripheral selectedPeripheral` field 
- add `IGPIOReceiver` interface and a method that would look like this:
```
public void OnGPIO(int number, bool value)
{
    if(!value)
    {
        selectedPeripheral = null;
    }
    else
    {
        if(!TryGetByAddress(number, out selectedPeripheral))
        {
             // log an error that the peripheral is missing
        }
    }
}
```
- switch all usages of `RegisteredPeripheral` to `selectedPeripheral` (remember to check for null values)
- in the repl file, connect the gpio signal to spi (!). 

```
gpio:
    6 -> spi@0  // this number after `@` has to be the same as the connection number you use to connect flash to spi
```

mkschreder, 2022-12-12:
@PiotrZierhoffer @maximevince I have implemented CS pin in this driver: https://github.com/swedishembedded/sdk/blob/main/renode/drivers/gpio/Mcp23S17.cs

maximevince, 2022-12-13:
Awesome, I have yet to try it...

closed 2023-06-19 by PiotrZierhoffer