renode/renode #396: STM32F072RB RCC Mocked Incorrectly

Seth10001, 2022-10-19:
### Description

I'm helping [Solar Gators](https://ufsolargators.org) create some system tests for their Sun Rider car. They've got a few boards on the car that talk over CAN and an automated system test for key requirements would be helpful for regression.

I've got our telemetry board running [with some modified code](https://github.com/John-Carr/TelemetryFirmware/commit/4a80c39d3bb73f3482e3598cbb8d6aed5db69d7a), the diff for main.c isn't great so I'll point out the two lines I changed:

- [SystemClock_Config](https://github.com/John-Carr/TelemetryFirmware/blob/4a80c39d3bb73f3482e3598cbb8d6aed5db69d7a/Core/Src/main.c#L104)
- [MX_CAN_Init](https://github.com/John-Carr/TelemetryFirmware/blob/4a80c39d3bb73f3482e3598cbb8d6aed5db69d7a/Core/Src/main.c#L112)

For `SystemClock_Config` it seems like the RCC mock is returning incorrect data for our configuration. When I run renode [one of these lines don't return HAL_OK](https://github.com/John-Carr/TelemetryFirmware/blob/4a80c39d3bb73f3482e3598cbb8d6aed5db69d7a/Core/Src/main.c#L185-L209).

Here's the stack trace of interest:
```
08:37:05.6108 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EDE
08:37:05.6109 [NOISY] rcc: READ on FLIPFLOP at 0x34, value 0x0
08:37:05.6110 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EEA
08:37:05.6113 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002FB4
08:37:05.6117 [INFO] cpu: Entering function SystemClock_Config at 0x8000592
08:37:05.6119 [INFO] cpu: Entering function SystemClock_Config at 0x8000596
08:37:05.6123 [INFO] cpu: Entering function Error_Handler (entry) at 0x80008E4
```

My guess [is the RCC mock](https://github.com/renode/renode/blob/master/platforms/cpus/stm32f0.repl#L151-L162) is doing something it isn't supposed to be.

**Questions**
- Can I override the rcc mock? When I've tried before I got an error.
- Is the mock correct?

### Expected behaviour

The mock should return correct data and not fail.

### Environment

OS: Mac

.repl file: https://github.com/Solar-Gators/sunrider-system-testing/blob/main/test.resc

### Do you plan to address this issue and file a PR?

I would love to!

### Stack Trace
```
08:37:05.5875 [INFO] cpu: Entering function SystemClock_Config (entry) at 0x8000518
08:37:05.5875 [INFO] cpu: Entering function memset (entry) at 0x800BB02
08:37:05.5875 [INFO] cpu: Entering function memset at 0x800BB0C
//... lots of memset's we don't care about
08:37:05.5979 [INFO] cpu: Entering function SystemClock_Config at 0x800054C
08:37:05.5979 [INFO] cpu: Entering function HAL_RCC_OscConfig (entry) at 0x8002714
08:37:05.5980 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002728
08:37:05.5980 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002732
08:37:05.5980 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002852
08:37:05.5980 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x800285C
08:37:05.5983 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002938
08:37:05.5989 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x80029B0
08:37:05.5991 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x80029BA
08:37:05.5992 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002B28
08:37:05.5994 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002BFA
08:37:05.5995 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C00
08:37:05.6002 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C04
08:37:05.6007 [NOISY] rcc: READ on FLIPFLOP at 0x4, value 0xfffffff8
08:37:05.6011 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C10
08:37:05.6012 [NOISY] rcc: READ on FLIPFLOP at 0x4, value 0x0
08:37:05.6016 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C46
08:37:05.6019 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C4E
08:37:05.6020 [NOISY] rcc: READ on FLIPFLOP at 0x34, value 0xffffffff
08:37:05.6021 [NOISY] rcc: WRITE on FLIPFLOP at 0x34, value 0xffffffff
08:37:05.6027 [INFO] cpu: Entering function HAL_GetTick (entry) at 0x8001600
08:37:05.6029 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C60
08:37:05.6032 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C78
08:37:05.6033 [NOISY] rcc: READ on FLIPFLOP at 0x34, value 0x0
08:37:05.6034 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C66
08:37:05.6035 [INFO] cpu: Entering function HAL_GetTick (entry) at 0x8001600
08:37:05.6040 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C6A
08:37:05.6043 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C78
08:37:05.6045 [NOISY] rcc: READ on FLIPFLOP at 0x34, value 0xffffffff
08:37:05.6048 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C84
08:37:05.6052 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002CBE
08:37:05.6054 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002CC6
08:37:05.6057 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002E12
08:37:05.6058 [INFO] cpu: Entering function SystemClock_Config at 0x8000566
08:37:05.6065 [INFO] cpu: Entering function SystemClock_Config at 0x800056E
08:37:05.6066 [INFO] cpu: Entering function HAL_RCC_ClockConfig (entry) at 0x8002E20
08:37:05.6073 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002E34
08:37:05.6083 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002E42
08:37:05.6084 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002E66
08:37:05.6086 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002E70
08:37:05.6093 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002E7A
08:37:05.6095 [NOISY] rcc: READ on FLIPFLOP at 0x4, value 0x0
08:37:05.6096 [NOISY] rcc: WRITE on FLIPFLOP at 0x4, value 0x700
08:37:05.6098 [NOISY] rcc: READ on FLIPFLOP at 0x4, value 0xfffffff8
08:37:05.6100 [NOISY] rcc: WRITE on FLIPFLOP at 0x4, value 0xffffff08
08:37:05.6103 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EA6
08:37:05.6104 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EBE
08:37:05.6107 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002ED6
08:37:05.6108 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EDE
08:37:05.6109 [NOISY] rcc: READ on FLIPFLOP at 0x34, value 0x0
08:37:05.6110 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002EEA
08:37:05.6113 [INFO] cpu: Entering function HAL_RCC_ClockConfig at 0x8002FB4
08:37:05.6117 [INFO] cpu: Entering function SystemClock_Config at 0x8000592
08:37:05.6119 [INFO] cpu: Entering function SystemClock_Config at 0x8000596
08:37:05.6123 [INFO] cpu: Entering function Error_Handler (entry) at 0x80008E4
08:37:05.6130 [NOISY] cpu: Loop to itself detected
08:37:05.6134 [INFO] cpu: Entering function Error_Handler at 0x80008EA
```

Seth10001, 2022-10-20:
I switch the rcc to
```
rcc: Miscellaneous.STM32F4_RCC @ sysbus 0x40021000
    rtcPeripheral: rtc
```

And I'm getting this error:
```
20:25:15.2498 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002BD8
20:25:15.2498 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002BE4
20:25:15.2499 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C1A
20:25:15.2499 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C22
20:25:15.2499 [WARNING] rcc: Unhandled write to offset 0x34. Unhandled bits: [16] when writing value 0x10000. Tags: RESERVED (0x100).
20:25:15.2499 [INFO] cpu: Entering function HAL_GetTick (entry) at 0x80015D4
20:25:15.2499 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C34
20:25:15.2500 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C4C
20:25:15.2500 [INFO] cpu: Entering function HAL_RCC_OscConfig at 0x8002C3A
20:25:15.2500 [INFO] cpu: Entering function HAL_GetTick (entry) at 0x80015D4
```

PiotrZierhoffer, 2022-11-15:
Hi @Seth10001 

The RCC mock is just a mock - it was created for some specific application which worked with the current implementation, but it's definitely not a fully fledged model.

`STM32F4_RCC` is also a mock, but a bit more advanced. We also have `STM32L0_RCC`, slightly newer and more advanced, but a mock nevertheless.

The reason why we mock these is that RCC peripherals are not very interesting when we pass the initialization phase, where it usually boils down to "let's read a value we expect or we just written to it". We very rarely use the values written to RCC anyway, as things like frequencies are often configured in REPL files.

The warning you get about unhandled write is often not an issue - it just tells you that a specific register is not implemented, and if you're not reading from it you should be ok.
In your particular case it says you try to write to a reserved field - but you use RCC from another platform, so it's not really unexpected.

Long story short I would suggest:
- use STM32F4_RCC or STM32L0_RCC
- if your application doesn't break on rcc, just go with it, even if you have warnings
- if you want, feel free to create an F0 version of this peripheral, probably based on L0, and implement the registers you really care about.

PiotrZierhoffer, 2023-01-10:
We are currently not working on other RCC models.

Please let us know if the L0 model helped. I'm closing the issue for now, but feel free to comment if you have any other concerns in this topic.

closed 2023-01-10 by PiotrZierhoffer

Seth10001, 2023-05-29:
Hey @PiotrZierhoffer, I tried `STM32L0_RCC` by updating the `cpus/stm32f072.repl` rcc definition to the following:

```
rcc: Miscellaneous.STM32L0_RCC @ sysbus 0x40021000
    rtc: rtc
    lptimer: lptim1

lptim1: Timers.STM32L0_LpTimer @ sysbus 0x40007C00
    frequency: 0x1000000
    -> exti@2
```

After [running our tests](https://github.com/Solar-Gators/Sunrider-Firmware/blob/be91c92437096235a2f849cd7c186321f0ab4f9b/telemetry-firmware/tests/telemetry.robot) I am getting this error: 

```
08:33:10.4172 [WARNING] rcc: Unhandled write to offset 0x1C. Unhandled bits: [4] when writing value 0x10. Tags: IOPERST (0x1).
```

The `L0` mock might not match match `F0` exactly. Our tests do not depend on rcc so commenting this out for testing is working for us