renode/renode #397: GDB: `compare-sections` always reports failures

xobs, 2022-10-20:
### Description

The gdb command `compare-sections` can be used to determine if a program image in memory matches the image on disk. It does this by asking the host to calcualte a CRC of various ranges, which it then compares against its own hash.

Renode appears to report that it supports this command, however it is always returning a CRC of "1".

### Expected behaviour

Renode should either correctly calculate the CRC, or report an error if it is not supported.

### How to reproduce?

1. Load any ELF file into RAM.
2. Run `compare-sections`

### Environment

- OS: Windows
- Renode version (with commit SHA): 72c88bbf46903866c1b21365310637dfc82ae4b5

### Additional information

Renode is reporting a CRC of "1":

```
(gdb) compare-sections
Sending packet: $qCRC:10100,60dc#ce...Ack
Packet received: QC1
Section .rodata, range 0x10100 -- 0x161dc: MIS-MATCHED!
Sending packet: $qCRC:161dc,8a4#db...Ack
Packet received: QC1
Section .eh_frame_hdr, range 0x161dc -- 0x16a80: MIS-MATCHED!
Sending packet: $qCRC:16a80,1dd8#10...Ack
Packet received: QC1
Section .eh_frame, range 0x16a80 -- 0x18858: MIS-MATCHED!
Sending packet: $qCRC:19858,2454c#f0...Ack
Packet received: QC1
Section .text, range 0x19858 -- 0x3dda4: MIS-MATCHED!
Sending packet: $qCRC:3eda8,44#ac...Ack
Packet received: QC1
Section .sdata, range 0x3eda8 -- 0x3edec: MIS-MATCHED!
warning: One or more sections of the target image does not match
the loaded file

(gdb)
```

### Do you plan to address this issue and file a PR?

I'm not currently planning on creating a PR for this, though it does seem like it could be a Good First Issue.

mateusz-holenko, 2023-01-03:
Hi @xobs, thanks for reporting the issue.

In 7aefb555588fc9a968f0e9a015309c2997150247 we've reworked GDB packets mnemonics parsing logic nad added support for the `qCRC` command - it should now report proper results.

Could you verify if this works for you?

PiotrZierhoffer, 2023-02-24:
Closing this now, as discussed with @xobs

closed 2023-02-24 by PiotrZierhoffer