renode/renode #398: [Support request] gdb not working when using the load gdb command in k64 platform.

waltergallegog, 2022-10-21:
I'm using renode and the **k64f** platform.
I'm using a toy example from **zephyr** to use the UART, based on [this post](https://www.zephyrproject.org/developing-zephyr-rtos-embedded-applications-on-platformio-and-simulating-on-antmicro-renode/)

I'm able to run gdb using the following steps:

1 - Open renode monitor, create a machine and include the platform `@platforms/cpus/nxp-k6xf.repl`
2 - Load my debug .elf firmware using `sysbus LoadELF @firmware.elf`
3 - start a gdb server with `machine StartGdbServer 3333`
4 - Open gdb with the debug firmware  `./arm-none-eabi-gdb firmware.ef`
5 - Connect to the server `target remote :3333`
6 - start simulation with `monitor start` and `continue`

However, if instead of loading the firmware with `sysbus LoadELF`, I do it with the `load` command of gdb, the simulation does not work. (I don't see anything in my UART, the program seems stuck) here the steps:

1 - Open renode monitor, create a machine and include the platform `@platforms/cpus/nxp-k6xf.repl`
2 - start a gdb server with `machine StartGdbServer 3333`
3 - Open gdb with the debug firmware  `./arm-none-eabi-gdb firmware.ef`
4 - Connect to the server `(gdb) target remote :3333`
5 - Load the debug firmware with `(gdb) load`
6 - start simulation with `(gdb) monitor start` and `(gdb) continue`

So perhaps the load command of gdb is loading the elf firmware to a different address?
Where should I start looking at the issue? 
Is there any renode command that I can use to understand where the.elf is loaded in both cases?

The problem is not present when using the platform **sifive-fe310.repl**. (with the corresponding `riscv64-unknown-elf-gdb`)

Thanks for your support.

PiotrZierhoffer, 2022-11-15:
Hi @waltergallegog 

We can reproduce a similar issue on our side and we're currently investigating the possible cause.

mateusz-holenko, 2022-12-16:
Hi @waltergallegog,
it turned out to be related to virtual address translation in the GDB layer and should be fixed now in: https://github.com/renode/renode/commit/a5b6a02347681dfff5b0f83f9dd86b9d38439aa7.

Could you check if that resolves the problem on your side?

waltergallegog, 2022-12-17:
Hi @mateusz-holenko 
Thanks for the fix.
I have built the indicted commit and updated renode. I confirm that now the load command works.

Should I close this issue or will you handle it with a PR?

mateusz-holenko, 2022-12-19:
Thanks @waltergallegog, the fix is already on master so I'm closing this issue.

closed 2022-12-19 by mateusz-holenko