renode/renode #399: STM32L072 sysbus.usart5 "Peripheral not found"

sktpin, 2022-10-21:
Thanks for taking the time to file an issue for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.rst) first.

### Description

1st attempt at using this software, not sure I'm not justz doing it wrong.
Followed this tutorial: https://interrupt.memfault.com/blog/intro-to-renode
Except I did not use his board or binary, I created a new machine:
- mach add "mymachine"
- machine LoadPlatformDescription @platforms/cpus/stm32l072.repl
- sysbus LoadBinary @C:\path\to\project\Debug\myimage.bin 0x8000000

Then I was at the point in the tutorial to add an analyzer for the debug uart (or, uSart in my case). He uses UART2, from his description I should have been able to do this:
- showAnalyzer sysbus.usart5

But, apparently not, as I got: 'Peripheral not found: sysbus.usart5'

### Expected behaviour

According to the tutorial, no error message.

### How to reproduce?

In a PuTTY shell with a local Telnet session to the Renode monitor running with "-P 1234",  issue the commands:

1. mach add "mymachine"
2. machine LoadPlatformDescription @platforms/cpus/stm32l072.repl
3. sysbus LoadBinary @C:\path\to\project\Debug\myimage.bin 0x8000000
4. showAnalyzer sysbus.usart5

Where myimage.bin is built, with STM32CubeIDE, for the STM32L072CZ device, and works on real hardware.

I use PuTTY to have my local keyboard mapping working.

### Environment

- Windows 10
- Renode 1.13.2 - 2022.10.03

### Additional information

Please add all the information you feel is relevant but not clearly visible in the reproduction repo or otherwise.

### Do you plan to address this issue and file a PR?

No, I just started trying this whole thing, no idea what's going on

sktpin, 2022-10-24:
Ok, I looked a bit in the .repl files and tried to add the missing stuff to the stm32l071 file, which the l072 one is _using_ as a base.
So I added the missing uarts from another template, changed the nvic numbers according to the refman and my hopefully correct idea about what the repl files are doing there. Though I now saw that some of the uarts refer to nvic@number whereas others do to e.g. "nvicInput14@1", I wonder what that means.

But when loading that device, I'm getting an error seemingly unrelated to what I changed:

>  [ERROR] Received GPIO signal on an unsupported port #2 (supported ports are 0 - 1). Please check the platform configuration

And when I start my executable, I get:

> [ERROR] cpu0: PC does not lay in memory or PC and SP are equal to zero. CPU was halted

The binary works on real hardware, so I'm kinda lost here.

[stm32l071.repl.txt](https://github.com/renode/renode/files/9851251/stm32l071.repl.txt)

sktpin, 2022-10-26:
I simplified my program by a lot and also a restart of Renode seemed to be better than trying out different things & reloading stuff - it does run a bit further now and then the Renode Monitor prints out a warning that Flash:ACR unimplemented register is being accessed. This is in the _SystemClock_Config_() of the STM32CubeIDE generated init code, expecting to see a change in that register.
This is such an essential function at startup, and I found no single STM32 model in _platforms\cpus_ where this register is implemented, that I wonder about how anyone has run anything successfully that was generated with ST's default IDE.

Does anyone have a pointer for me how one would implement such a register?
Is this just adding some lines in the .repl file for my MCU (say, if a fixed value suffices, for now), or implementing some extra C# module?
I could, of course, just disable these things in the program and test only the bare things, but then I might as well have gone the QEMU route, kinda defeats the purpose. I _was_ impressed by the statement "use the same binary like on the real hardware".

shareefj, 2022-11-07:
Hi, just another newb user commenting...

- I found that any changes made to `repl` files require the tool to be restarted.  Things such as the virtual UART monitors hang on a restart so it's kind of useless
- a lot of the `repl` files don't implement the full peripheral set.  I'm assuming that this is just because they're a work in progress and no one has implemented them yet.  For example, I had to create my own `stm32f4.repl` which included the missing `ccm` memory as defined in those MemFault blog posts. I'd expect PR's fixing this would be welcome.
- A lot of the CSR registers are just not implemented because they are of no use in Renode, your clock registers for example.  Anything that isn't implemented normally returns a warning in the console so that you can either fix the code or implement any missing register.

In terms of your hang when accessing an unimplemented register, you probably need to ifdef out that section of code.  Renode doesn't model things like the clock generation hardware as it doesn't make any sense for an emulated CPU.  What is that code trying to do?  Setting flash wait states depending on clock frequency?

Perhaps someone from Renode will be along to describe how best to fix this but my understanding is you can't run this type of code on a Renode model.

If you really have to model this then I think you can either use SVD or copy/edit an existing C# model and implement the registers.

mateusz-holenko, 2022-11-15:
> But, apparently not, as I got: 'Peripheral not found: sysbus.usart5'

@sktpin, some time ago we've extended the platform description with new peripherals (it now includes `usart5`) so please check the newest package (could be obtained from https://builds.renode.io)

> Though I now saw that some of the uarts refer to nvic@number whereas others do to e.g. "nvicInput14@1", I wonder what that means.

This is related to the fact that it's possible for multiple peripherals to be connected to the some NVIC input and we model it in Renode using a `CombinedInput` entry in the repl file (`nvicInput14` is the name of this entry in this context).

> warnings that Flash:ACR unimplemented register is being accessed

In fact this should already be fixed by 10a863f, could you verify it works for you in the newest Renode?

sktpin, 2022-11-21:
> In terms of your hang when accessing an unimplemented register, you probably need to ifdef out that section of code. Renode doesn't model things like the clock generation hardware as it doesn't make any sense for an emulated CPU.

This is in direct contradiction to one core statement on Renode's website, of one of the goals of the framework being that you do **not change the firmware** image between real hardware vs. emulation. Which is a sensible goal, as you'd not be testing the real firmware otherwise.

> Perhaps someone from Renode will be along to describe how best to fix this but my understanding is you can't run this type of code on a Renode model.

That, combined with the "don't change firmware" idea, would mean that basically all but the most trivial "hello world" MCU firmware were no good match for this framework. That can't  be right.

mateusz-holenko, 2022-11-22:
> I found that any changes made to repl files require the tool to be restarted. Things such as the virtual UART monitors hang on a restart so it's kind of useless

Restart is not required, but you need to recreate the machine. Just issue the `Clear` command and load your platform once again. If anything hangs anywhere this is an issue, but we’d need to know more details.

> a lot of the repl files don't implement the full peripheral set. I'm assuming that this is just because they're a work in progress and no one has implemented them yet. For example, I had to create [...]

We usually implement platforms to the extent we need for some specific use case. Their feature set grows over time, and indeed we welcome PRs with additional blocks if we missed something.

> In terms of your hang when accessing an unimplemented register, you probably need to ifdef out that section of code.

Ifdefing is always the least preferred way of solving these issues. It’s often enough to add a stub model or even a Python peripheral to mock the behavior. We have a few pydevs defined here: https://github.com/renode/renode/tree/master/scripts/pydev - you can scan the platform files for these filenames and you will find usage examples.

> Renode doesn't model things like the clock generation hardware as it doesn't make any sense for an emulated CPU.

Renode comes with many features that allow for fast&easy prototyping - this means that in most cases you don't need to support the whole platform in order to simulate it.
Clock controllers are often a good example here - we define frequencies of clocks in repl so as long as lack of RCC (which in fact means that all writes to this peripheral will be ignored and all reads from it will return 0) doesn't cause SW to loop indefinitely (e.g., waiting for some bit to be set) it's not necessary to be included in the platform.

This doesn't mean though that you can't model RCC in Renode (in fact there already are clock controller models for selected platforms available). It's just a matter of how precise the platform model must be in order to successfully run your simulation (and this varies from binary to binary).

> my understanding is you can't run this type of code on a Renode model

Renode is meant to simulate actual HW and run unmodified software of any type. As mentioned above it's just a matter of how detailed the simulation must be for the particular use-case.

> This is in direct contradiction to one core statement on Renode's website, of one of the goals of the framework being that you do not change the firmware image between real hardware vs. emulation

We definitely believe this is possible - as I mentioned above, oftentimes a mock of a peripheral is enough to move forward, without additional changes to your software.