renode/renode #400: Use vmnet on macOs for HostInterface

xobs, 2022-10-25:
### Description

Apple has deprecated kernel extensions and may remove them at some point, so the existing `tap.kext` approach for attaching an emulated network to a physical target has a shelf life.

One approach is to use the virtualization framework to create a virtual network device. `vmnet.Framework` is the preferred approach for making virtual network devices that work with a virtualized host. This framework can be used in isolation to create a network interface and send Ethernet frames. This does not require any additional features other than (1) running as root or (2) giving the program a specific Entitlement.

### Usage example

An example of how to use this API is available at https://github.com/hamishcoleman/macos-vmnet/blob/master/simple.c#L110-L287

This approach is also taken by qemu, with their patch added at https://lists.gnu.org/archive/html/qemu-devel/2021-02/msg04637.html

### Additional information

I'm not sure how to access macOs `.Framework` files from .NET, so that may be a big hurdle. However, the approach seems to rely entirely on C API calls, so if .NET has the ability to call C functions then it should work just fine.

### Do you plan to address this issue and file a PR?

Not currently.

PiotrZierhoffer, 2022-10-25:
Hi @xobs , this is very interesting! The kext approach is difficult to manage and of course quite deprecated already, so an alternative makes sense.

I'm not sure when will we be able to try this, but it sounds like a reasonable approach.

keys-i, 2024-11-19:
Could we use system extension?