renode/renode #401: I am new to renode but unable to understand this clearly. How it simulate the peripherals any one please explain.

Arunprasad657, 2022-11-02:
Thanks for taking the time to submit a feature request for Renode!

Please familiarize yourself with our [Contributing guidelines](https://github.com/renode/renode/blob/master/CONTRIBUTING.rst) first.

### Description

A clear and concise description of what you expect to achieve.

### Usage example

Provide information how do you envision the feature to be used and what usage scenarios could be supported by this feature.

### Additional information

Please add all information you feel to be relevant.

### Do you plan to address this issue and file a PR?

Please let us know if you want to work on this yourself or need support.

mkschreder, 2022-12-14:
Peripherals are just memory mapped devices. It implements an api from simulated memory accesses to c#

PiotrZierhoffer, 2023-01-24:
Hi @Arunprasad657 

You question is quite vague, it's difficult to answer precisely. Let me direct you to our documentation on that matter: https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html

You can find our peripheral implementations here: https://github.com/renode/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals

Look for a simple uart and this will, hopefully, give you understanding of the modelling process.

closed 2023-01-24 by PiotrZierhoffer