renode/renode #402: build failed on MacOS M1

0xkelvin, 2022-11-09:
Hell Renode, 

i was able to compile Renode successfully at the first try.
after that, i installed Mono IDE ( i intended to use it for debug C# code). but then after that everything is break. I am not able to compile anymore. 
below is the error log, something gone wrong in my environment : 

/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/SelectColorDialogBackend.cs(34,3): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(63,10): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/HeaderBoxGtk2.cs(49,53): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(68,10): error CS0246: The type or namespace name 'TreeModel' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/HeaderBox.cs(119,54): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(70,10): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(146,44): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(146,69): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/HeaderBox.cs(35,3): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(168,4): error CS0246: The type or namespace name 'GLib' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(152,44): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(152,75): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(152,100): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(210,4): error CS0246: The type or namespace name 'GLib' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(245,4): error CS0246: The type or namespace name 'GLib' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(289,4): error CS0246: The type or namespace name 'GLib' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(158,44): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(158,94): error CS0246: The type or namespace name 'Gdk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend.CellViews/CellViewBackend.cs(303,4): error CS0246: The type or namespace name 'GLib' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TableStoreBackend.cs(45,10): error CS0246: The type or namespace name 'TreeModel' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/WidgetBackend.cs(99,10): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/WidgetBackend.cs(110,10): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/TreeViewBackend.cs(377,54): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)
	/Users/kelvin/Ionzone/Gridania/renode/lib/xwt/Xwt.Gtk/Xwt.GtkBackend/WidgetBackend.cs(377,3): error CS0246: The type or namespace name 'Gtk' could not be found (are you missing a using directive or an assembly reference?)

0xkelvin, 2022-11-09:
one more information, i hope this help : 
[image]

PiotrZierhoffer, 2022-11-15:
Hi @0xkelvin 

Building for dotnet and mono are two different things, and I'd suggest using separate clones for that.

I assume you try to build for x86? Our translation libraries are not yet available for ARM cores.

Did you try first to build with --net switch?

What does ` gacutil -l | grep -i gtk` tell you?

0xkelvin, 2022-11-18:
oh, thanks for your answer. 
I actually try to build for my macbook M1.

PiotrZierhoffer, 2022-11-18:
This, in general, shouldn't work - our CPU simulation parts are x86_64-specific. I asked about x86, as I thought you might want to use Rosetta, M1 own binary translator.

We had reports of people running Renode under Rosetta, but this is not something we properly tested.

Anyway, my question about mixing build types holds - I believe this might be the source of the problem.

JoseClaudioSJr, 2022-12-26:
Recommended to use a Docker image with Colima. By command "colima start -c ${VAL1} -m ${VAL2} -a x86_64", it is possible to start an x86_64 environment on Apple Silicon and run Renode normally through the Docker Image.

rchtsang, 2023-08-23:
Don't know if this issue is supposed to be still open, but i'm running on Mac M1 Pro and I'm able to natively compile renode at tag v1.13.3 (`Renode, version 1.13.3.35686 (a72a1fa1-202308221949)
`)

haven't tried v1.14 yet, but just an fyi.

rchtsang, 2024-05-01:
just to clarify, did 1.15.0 add support for M1 builds? I am a bit confused since elsewhere it says "added support for AArch64/ARM64" for issues that sound like they're related to host architecture.

boyesm, 2024-06-02:
On first compile using the typical method, I had errors as well:

```
$ ./build.sh

...

Errors:

/Users/malcolm/Projects/OSS/renode/Renode.sln (default targets) ->
(Build target) ->
/Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/cores-arm64.csproj (default targets) ->
/Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/cores.template.csproj (CompileTranslationLib target) ->
/Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj (VerifyProperties;GenerateFlags;Compile;Link;Build) ->
(Compile target) ->

        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error : tlib/arch/arm64/mte_helper.c:232:51: warning: too many arguments in call to 'error_get_pretty'
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error :                               error_get_pretty(err));
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error : tlib/arch/arm64/mte_helper.c:232:47: warning: passing arguments to 'error_get_pretty' without a prototype is deprecated in all versions of C and is not supported in C2x [-Wdeprecated-non-prototype]
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error :                               error_get_pretty(err));
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error : tlib/arch/arm64/mte_helper.c:233:31: warning: too many arguments in call to 'error_free'
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error :                 error_free(err);
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error : tlib/arch/arm64/mte_helper.c:233:27: warning: passing arguments to 'error_free' without a prototype is deprecated in all versions of C and is not supported in C2x [-Wdeprecated-non-prototype]
        /Users/malcolm/Projects/OSS/renode/src/Infrastructure/src/Emulator/Cores/translate.cproj: error :                 error_free(err);

         6067 Warning(s)
         8 Error(s)
```

However, using a modified version of what @rchtsang did on 1.13.3, I was able to get it to build:

```
$ arch -x86_64 /bin/bash -c ./build.sh
... build fails, errors
$ arch -x86_64 /bin/bash -c ./build.sh
... build successful
```

My system/config:

```
$ uname -mrs
Darwin 23.2.0 arm64

$ mono --verison
Mono JIT compiler version 6.12.0.206 (2020-02/0cbf0e290c3 Tue Feb 13 17:04:08 EST 2024)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           
        SIGSEGV:       altstack
        Notification:  kqueue
        Architecture:  amd64
        Disabled:      none
        Misc:          softdebug 
        Interpreter:   yes
        LLVM:          yes(610)
        Suspend:       hybrid
        GC:            sgen (concurrent by default)

$ git show
commit f830e6345401c39fa685c35dd32ad214d6181bca (HEAD -> master, origin/master, origin/HEAD)
...
```

Unsure why this is the case, but it should work as an interim solution.

rchtsang, 2024-06-17:
any updates on supporting native build? tried building current master just now and still getting errors