renode/renode #403: PLIC interrupt still repeats infinitely after completion claiming

jhegr, 2022-11-14:
### Description

We use FreeRTOS running on a board with FPGA implementing RISC-V processor. The FPGA generates external interrupts driving main system clock (10 kHz). ISR of this interrupt increments a tick for the FreeRTOS scheduler.

Because the board is not ready yet, we use RENODE for its simulation. The external interrupts are simulated via a timer, set in a _repl_ file in this way:
```
cpu: CPU.RiscV32 @ sysbus
    cpuType: "rv32imafc"
    privilegeArchitecture: PrivilegeArchitecture.Priv1_09
    timeProvider: clint

plic: IRQControllers.PlatformLevelInterruptController @ sysbus 0x40000000
    [0-3] -> cpu@[8-11]
    numberOfSources: 32
    prioritiesEnabled : true

clint: IRQControllers.CoreLevelInterruptor  @ sysbus 0x44000000
    frequency: 50000000
    [0, 1] -> cpu@[3, 7]

timer0: Timers.MiV_CoreTimer @ sysbus 0x70003000
    -> plic@31
    // clockFrequency: 2560000 // For the Limit set to 0xFF, it's 10 kHz
    clockFrequency: 131070 // DEBUG: For the Limit set to 0xFFFF the clock is 0.5 s
```
and in the _resc_ file we enable the timer:
```
macro rtce_timer
"""
sysbus.timer0 Limit 0xFFFF
sysbus.timer0 Enabled True
"""
runMacro $rtce_timer
```
**And here is a problem we cannot figure out:**
From the first moment, we were searching why all system tasks are not running, but ISR is called. Later we found out that the external PLIC interrupt 31 is not completed after the ISR has been finished and the endless interrupting does not allow OS to work properly. The interrupt handler looks like this:
```
void handle_m_ext_interrupt(void)
{
    uint32_t int_num  = PLIC_ClaimIRQ();
    uint8_t disable = EXT_IRQ_KEEP_ENABLED;

    disable = ext_irq_handler_table[int_num]();

    PLIC_CompleteIRQ(int_num);

    if(EXT_IRQ_DISABLE == disable)
    {
        PLIC_DisableIRQ((IRQn_Type)int_num);
    }
}
```
But the interrupt is triggered infinitely. We have not found out how to claim completion, even from the RENODE monitor window. The timer triggers the first interrupt. From that moment the interrupts are still generated. We disabled the timer, but the triggering was still happening.

Does anybody know what we miss in the platform configuration? Our application does not use timers, do we need to add some interrupt finish acknowledge in the code (i.e., set some additional flag in some register address)?

We thank you for any help.

### Additional information

We use Renode version 1.11.0.8104.

### Do you plan to address this issue and file a PR?

No, we don't.

PiotrZierhoffer, 2022-11-15:
Hi @jhegr 

Few questions:

- have you tried the latest Renode version? You can grab one from https://builds.renode.io/
- you can verify if the reissuing of the interrupt is not triggered by the timer itself. Are you sure you clear the interrupt on the timer?

You can do it manually from the Monitor with `sysbus.timer0 WriteDoubleWord 0x10 1`.

In general we believe claiming interrupts on PLIC should work just fine.

You can inspect your interactions with peripherals with `sysbus LogAllPeripheralsAccess true` - you can also attach this log here, maybe it will be more obvious then.

jhegr, 2022-11-21:
I really thank you for your help. Writing **1** to the timer offset did the trick. We have it in our code, but we used wrong offsets. Unfortunately, I've ignored RENODE messages related to 'Unhandled write to offset 0x...' because there are more, probably not so important for our purpose. Now I'll be more careful.

PiotrZierhoffer, 2022-11-21:
Great to hear that.

Indeed, sometimes you may get many warnings in your log that are not super relevant, this depends on the platform completeness. Maybe this will help?

Can we close this issue?

jhegr, 2022-11-21:
Great! This will help. Thank you.
And yes, we can close this thread.

closed 2022-11-21 by PiotrZierhoffer