renode/renode #404: Use IF/ELSE syntax in Robot Framework

admorgan, 2022-11-14:
### Description

Robot Framework is now on release 6, which has deprecated some of the keywords used in Renodes Robot Framework
infrastructure. Specifically the Run Keyword If and Run Keyword Unless keywords are deprecated in favor of the IF/ELSE
syntax available. As this syntax is available in Renode's minimum supported version I would like to convert these features
as I am able to successfully utilize the Renode tools in my project utilizing Robot Framework 6.0.1.

### Do you plan to address this issue and file a PR?

I plan to submit a PR to address this feature request.

PiotrZierhoffer, 2022-11-15:
Thanks for the contribution @admorgan 

This looks reasonable, we'll look into merging this in.

Any other issues or benefits related to Robot 6 over 5?

admorgan, 2022-11-15:
RF is currently has a requirement on 4. We gained a TRY/EXCEPT in 5 that we have found simplifies a lot of our code. I also make use of variable files imported as modules. The modifications to rounding to match python also limits a weird set of errors that are very difficult to debug.

RF 6 adds support for python 3.11, which in testing in my environment is ~18% faster than 3.10. The enhanced tagging capabilities and improved listener features allow of a lot of flexibility in complicated tests. This can be especially important for machine under test style infrastructures. We use a lot of libraries and modules, the new keyword namespace feature allows for better keywords because you no longer have to ensure they are globally unique which can be complicated. I also appreciate the improved embedded arguments fixes. Nothing worse than calling the wrong keyword because one was being too greedy.

admorgan, 2022-12-02:
Anything I can do to assist with getting this accepted?

mateusz-holenko, 2023-01-05:
@admorgan Your PR has been merged upstream in: https://github.com/renode/renode/commit/01c9e5ce19110fe3b302f33b7ab5ef224a4b56f4. Thanks again for your contribution!

closed 2023-01-05 by mateusz-holenko