renode/renode #406: Is there a way to detect that the Monitor has terminated the Emulator's RequestReset?

gonbbb, 2022-11-15:
### Description

I was debugging the firmware and put together the following macro to restore the firmware to its initial state.

```
# example.repl
set resetAndStart
"""
monitor.Parse('machine RequestReset')
monitor.Parse('start')
"""

# monitor
(machine-0) runMacro $resetAndStart
```

However, `start` was issued without finishing the `machine RequestReset` process of the firmware, probably because the Monitor(CLI) and Emulator were not synchronised.

Expectation: firmware start from entry point.
Actual: previous programme state (e.g. program counter)

Is there a way for Monitor(CLI) to detect the end of `RequestReset` to solve this problem?

I would like to consider means other than `Thread.Sleep`.

e.g. 

https://github.com/renode/renode-infrastructure/blob/63702bc742f5247a691d19f128b5f16b3df32328/src/Emulator/Main/Core/Emulation.cs#L267

### Usage example

Repeatedly, when you want to return the state of the programme to its initial state.

As the current Renode takes a great deal of time to boot and load binaries, I thought this function would be useful when I wanted to make short work of it.

### Additional information

- OS: macOS Big Sur v11.4
- Renode: version 1.13.0.24782 

Thank you.

PiotrZierhoffer, 2022-11-15:
Hi @gonbbb 

I'm not sure I fully understand your steps here.

Let's start with your snippet. You are mixing two things here, Python and Monitor macro. To run `resetAndStart` you'd need to use `python $resetAndStart` - or am I misunderstanding something?

What you are, I believe, trying to achieve, is usually solved with the `reset` macro (but I believe you already have it in place).

To call it, just use `machine Reset`, not `RequestReset`.

You can define something like this:
```
macro test
"""
machine Reset
start
"""
```

`RequestReset` is an asynchronous version of `Reset`.

Please let me know if it helps!

gonbbb, 2022-11-15:
I was using macro incorrectly. This is exactly what I wanted to do.

> macro test
"""
machine Reset
start
"""

The source code convinced me.

> RequestReset is an asynchronous version of Reset.

The questions in this issue have been resolved and will be closed.
Thank you!

closed 2022-11-15 by gonbbb