renode/renode #407: Got issue when trying to debug Renode source with VSC and Omnisharp

eMagic1, 2022-11-16:
### Description

I try to debug Renode source code with Visual Studio code and Omnisharp on Ubuntu 20.04, with install mono-complete and mono developer. But when I start debug, the debug console show as below:
[image]

[image]

Please guide me how I can start debug
### Environment

Please, provide the following information:

- OS : Ubuntu 20.04 on Oracle Virtual Box
- Renode version (with commit SHA)  1.13.1.21690

MateuszKarlic, 2022-11-25:
Hi @eMagic1,

We are aware of the problem on Ubuntu 20.04, and we will look into resolving it, however it likely won't happen soon.
For now, as a workaround, you can manually start the debugger from a terminal with command `mono --debug --debugger-agent=transport=dt_socket,server=y,address=127.0.0.1:55555 output/bin/Debug/Renode.exe`. Afterwards, in `Run And Debug` pane, run target `Attach`, and you should be able to debug Renode.

PiotrZierhoffer, 2023-01-03:
@eMagic1 as you :+1: 'd the answer, I'll close the issue. We suspect it might be related to the Mono version that comes with this particular Ubuntu.

closed 2023-01-03 by PiotrZierhoffer