renode/renode #408: Request for STM32L452 support

dtran11, 2022-11-26:
Are there any plans to support STM32L4 series? thank you

PiotrZierhoffer, 2022-11-29:
Hi @dtran11 

We don't have specific plans for L4s, but please take a look at our Zephyr Dashboard: https://zephyr-dashboard.renode.io/

You will find L4 targets built and running sample Zephyr software.

[image]

If you click on the board name, you will be redirected to Renodepedia, and in the "Hardware" tab look for the "Renode dts2repl" button, it will give you the platform file usable by Renode.

[image]

dtran11, 2022-12-04:
> Hi @dtran11
>
> We don't have specific plans for L4s, but please take a look at our Zephyr Dashboard: https://zephyr-dashboard.renode.io/
>
> You will find L4 targets built and running sample Zephyr [...]

Thank you

PiotrZierhoffer, 2023-01-03:
I'll close the issue for now, feel free to open new ones if you have other questions

closed 2023-01-03 by PiotrZierhoffer