renode/renode #410: UART parsing and "Wait For Line On UART"

mkschreder, 2022-11-28:
### Description

What would be a good way to insert a protocol parser between TerminalTester and IUART?

Situation:

- There is binary protocol on UART
- I would like to implement a translation layer between "Wait For Line On Uart" and the actual uart output

The idea is to translate the binary data into text so that it can be expressed in robot script. I would like to take the text passed to "Wait For Line On Uart", convert it into intermediate format, take output of uart, convert that into intermediate format - and then use the existing TerminalTester to compare the two.

### Usage example

- Wait For Line On Uart  "{ 'json': 'data' }"
- Firmware prints binary data to uart
- Some kind of glue code parses uart data and prints out HEX data.
- The same code parses json data and converts it to HEX data
- Existing terminal tester basically gets equivalent of Wait For Line On Uart  "<hex data>"
- Result is that it becomes possible to wait for a particular binary message on uart while having all the benefits of current terminal tester that skips all data that is not part of the string we want to match.

What would be a good way to implement this? How can this be implemented in python?

PiotrZierhoffer, 2022-11-29:
Hi Martin

Not sure if I fully understand. Would you like to just be able to parse binary data by providing hex patterns? That should generally be quite easy, we just need a new keyword in `UARTKeywords.cs` and either a fork of the current `TerminalTester` or some improvements inside. I would say that it makes sense to have an explicitly separate keyword - current tester relies a lot on the notion of "lines", which does not make much sense in binary data.

`NetworkInterfaceTester` already has a concept of providing binary data as string of hex values. The tests using it are not super pretty, but, well, we hunt for binary data.

If you want to do more processing, the way I see it, we should use the `connector` based approach. You can take a look at `UARTHub`, it's a separate entity you can connect to and do whatever you want with the incoming data. It could also implement `IUART` to be able to attach the terminal tester to it. I haven't tested it, but I believe it would be possible with the current infrastructure.

mkschreder, 2022-11-30:
@PiotrZierhoffer what I would like to do is have an equivalent of "Wait For Line On Uart" but for binary data instead of lines. 

I'm not sure how I would use "connector" from robot script though? 

Could it be that the best approach in this case is to implement a separate uart tester for the custom data stream?