renode/renode #411: When loading more than one .elf file ...

sktpin, 2022-11-29:
There is probably no "bug", but it's also not a new feature - unless there are any side effects I'm not aware of. (my program crashes in Renode at some point with PC=0, but no idea whether that can be related).

### Description

Loading 2 ELF files: a bootloader and an app firmware.

One can already do that - but I'm not sure it's correct (what I do & expect).

### Usage example

I notice that if, in my .resc file, I load my bootloader ELF first, then the app ELF with sysbus LoadELF, the settings for that seem to be overwritten by the last load, and it guesses the VTOR of the app - so I swapped the order of ELF loading in my .resc script and it guesses the bootloader one, which needs to start.

Are there any side effects to be expected from doing that, or is that "the way to do it"?
Btw. why is it called VTOR "_guessing_"? That word does not instill confidence ;)

PiotrZierhoffer, 2022-11-29:
Well, it is guessing :) We are looking for the lowest ELF section we loaded, as this is often (but not always) a place where vector table is linked in.

You can set the VTOR yourself, to make your script fully resilient:

```
cpu VectorTableOffset 0xYourValue
```

But VTOR should not be affected by elf loading order, it's a bug.

PiotrZierhoffer, 2022-12-05:
I was too hasty to mark it as a bug. This is not Cortex-M specific, but works consistently across architectures.

So yes, the last ELF you load will set up your CPU starting point.

This definitely needs to be documented better.

mkschreder, 2022-12-12:
I typically directly load PC and SP from first two bytes of image or bootloader. 

For example, for MCUboot case I use this to boot the bootloader:
```
sysbus.cpu SP `sysbus.flash ReadDoubleWord 0x0`
sysbus.cpu PC `sysbus.flash ReadDoubleWord 0x4`
```

This is because PC and SP are always first words of the vector table (which is why Piotr's solution is probably a bit more elegant - but as you can see there is a lot of flexibility about how to start your CPU).

PiotrZierhoffer, 2023-01-08:
We have added a relevant note to the docs, closing

closed 2023-01-08 by PiotrZierhoffer