renode/renode #412: sysbus.cpu - No such command or device

sktpin, 2022-11-29:
### Description

Following this: [https://renode.readthedocs.io/en/latest/basic/logger.html](url)
-> _Creating a trace of the execution_
you're supposed to enter, with an open machine:

>  sysbus.cpu LogFunctionNames true

upon which I get

> No such command or device: sysbus.cpu

### Expected behaviour

Supposedly, that turns on logging of functions

### How to reproduce?

On the freshly started Renode monitor:

> (monitor) mach create
> (machine-0) machine LoadPlatformDescription @platforms/cpus/stm32l071.repl
> (machine-0) sysbus.cpu LogFunctionNames true
> No such command or device: sysbus.cpu
> 

### Environment

- Windows 10
- Renode, version 1.13.2.6137 (1a40d782-202211170323)

### Additional information

This was  true for **stm32l071**.repl but _not_ e.g. stm32g0.repl

### Do you plan to address this issue and file a PR?

*IF* I see how... I'll look into it, but I'm the noob!

PiotrZierhoffer, 2022-11-29:
Hi @sktpin,

this is quite strange. Have you built Renode yourself or is it a package? Is your repo/installation directory clean?

I can't reproduce it - the screen below shows the same commit, and it works just fine.

[image]

sktpin, 2022-11-30:
D'OH. I just copied the line from the description 1:1, didn't realise I had to number the CPU.

closed 2022-11-30 by sktpin