renode/renode #413: logFile output not the same as console output

mkschreder, 2022-11-30:
### Description

Is there a standard way to display function name profiling data? I was thinking to add a python script that can extract function name data from the log when "LogFunctionNames" is enabled, but I ran into a problem. 

The problem is that the log file does not contain precise timing data like the console output. I could of course run renode through "tee" or something and then use the output of that as raw input to a python script, but isn't this a bug though:

[image]

Shouldn't the outputs actually be identical? How come there no millisecond data in the log file?

### Expected behaviour

logFile output should have millisecond data.

PiotrZierhoffer, 2022-12-01:
Fixed in c87c852

PiotrZierhoffer, 2023-01-03:
If you still see there's an issue on your side, please reopen/file a new issue. Closing for now.

closed 2023-01-03 by PiotrZierhoffer