renode/renode #414: STM32L071.repl uses wrong ADC

sktpin, 2022-12-01:
### Description

The cpus/stm32l071.repl  contains:

> adc1: Analog.STM32_ADC @ sysbus 0x40012400
>     IRQ->nvic@12

This seems to be the implementation:
[STM32_ADC.cs](https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Analog/STM32_ADC.cs)

Some registers have wrong offsets, th enumber of overall registers is wrong, in the above implementation, vs. what's in the STM32L071. E.g. WatchdogLowerThreshold is assigned 0x28, but in the stm32071, that's ADC_CHSELR.
Also, this MCU's ADC only has 18 channels, not 19. (Looks like it's a bit simpler ADC)

[reference manual](https://www.st.com/resource/en/reference_manual/rm0377-ultralowpower-stm32l0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf)

See p. 327, section 13.12 ADC register map.

### Expected behaviour
### How to reproduce?
### Environment

- Renode version 1.13.2.6137 (1a40d782-202211170323)

### Additional information
### Do you plan to address this issue and file a PR?
Not sure I know enough. I'll look at it.

mateusz-holenko, 2023-01-03:
Hi @sktpin, thanks for reporting an issue.

In Renode there is another model of ADC for STM32 - namely https://github.com/renode/renode-infrastructure/blob/master/src/Emulator/Peripherals/Peripherals/Analog/STM32F0_ADC.cs.
From the registers layout perspective it seems to be compatible with the STM32L071 platform (although it also supports 19 channels).

Could you check if this model works for you? If so we can make it more generic (configurable number of channels) and fix the repl file.

nandojve, 2023-10-24:
Hello @mateusz-holenko ,

I had same issue and tried your solution where:

```
adc1: Analog.STM32F0_ADC @ sysbus 0x40012400
    referenceVoltage: 3.3
    externalEventFrequency: 1000
    IRQ->nvic@12
```

```
west build -b nucleo_l073rz samples/drivers/adc

...

15:49:32.3650 [INFO] Loaded monitor commands from: /home/gfbudke/renode/scripts/monitor.py
15:49:32.3967 [INFO] Including script: /home/gfbudke/zephyr/zephyr/boards/arm/nucleo_l073rz/support/nucleo_l073rz.resc
15:49:32.4088 [INFO] System bus created.
15:49:32.9202 [WARNING] RTC not passed in the RCC constructor. Changes to the real-time clock will be ignored
15:49:32.9202 [WARNING] Lptimer not passed in the RCC constructor. Changes to the low-power timer clock will be ignored
15:49:33.0903 [INFO] sysbus: Loading segment of 17312 bytes length at 0x8000000.
15:49:33.0985 [INFO] sysbus: Loading segment of 290 bytes length at 0x80043A0.
15:49:33.0985 [INFO] sysbus: Loading segment of 4 bytes length at 0x80044C2.
15:49:33.0985 [INFO] sysbus: Loading segment of 3936 bytes length at 0x20000128.
15:49:33.3555 [INFO] cpu0: Setting initial values: PC = 0x80007C5, SP = 0x20001088.
15:49:33.3567 [INFO] nucleo_l073rz: Machine started.
15:49:33.3937 [WARNING] sysbus: [cpu0: 0x80005D8] ReadDoubleWord from non existing peripheral at 0x40007000.
15:49:33.3939 [WARNING] sysbus: [cpu0: 0x80005D8] WriteDoubleWord to non existing peripheral at 0x40007000, value 0x800.
15:49:33.3941 [WARNING] sysbus: [cpu0: 0x80020FA] ReadDoubleWord from non existing peripheral at 0x40007000.
15:49:33.3942 [WARNING] sysbus: [cpu0: 0x8002130] ReadDoubleWord from non existing peripheral at 0x40007000.
15:49:33.3942 [WARNING] sysbus: [cpu0: 0x800210E] ReadDoubleWord from non existing peripheral at 0x40022000.
15:49:33.3942 [WARNING] sysbus: [cpu0: 0x800210E] WriteDoubleWord to non existing peripheral at 0x40022000, value 0x1.
15:49:33.3942 [WARNING] sysbus: [cpu0: 0x800210E] ReadDoubleWord from non existing peripheral at 0x40022000.
15:49:33.3943 [WARNING] sysbus: [cpu0: 0x8002128] ReadDoubleWord from non existing peripheral at 0x40022000.
15:49:33.3945 [WARNING] sysbus: [cpu0: 0x800146E] ReadDoubleWord from non existing peripheral at 0x40010020.
15:49:33.3945 [WARNING] sysbus: [cpu0: 0x800146E] WriteDoubleWord to non existing peripheral at 0x40010020, value 0x2000.
15:49:33.3960 [WARNING] sysbus: [cpu0: 0x800062C] ReadDoubleWord from non existing peripheral at 0x40015804.
15:49:33.3960 [WARNING] sysbus: [cpu0: 0x800062C] WriteDoubleWord to non existing peripheral at 0x40015804, value 0x0.
15:49:33.4028 [WARNING] rcc: Unhandled write to offset 0x28. Unhandled bits: [17] when writing value 0x20000. Tags: USART2RST (0x1).
15:49:33.4050 [WARNING] adc1: Unhandled write to offset 0x10. Unhandled bits: [31] when writing value 0x80000000. Tags: CKMODE (0x2).
15:49:33.4052 [WARNING] adc1: Unhandled write to offset 0x8. Unhandled bits: [28] when writing value 0x10000000. Tags: RESERVED (0x800000).
15:49:33.4097 [WARNING] adc1: Unhandled write to offset 0x8. Unhandled bits: [31] when writing value 0x80000000. Tags: ADCAL (0x1).
15:49:33.4124 [INFO] usart2: [host: 0.43s (+0.43s)|virt: 0.1ms (+0.1ms)] *** Booting Zephyr OS build zephyr-v3.5.0-420-g01aa8004644b ***
15:49:33.4128 [INFO] usart2: [host: 0.43s (+0.48ms)|virt: 0.13ms (+33?s)] ADC reading[0]:
```

otavio, 2023-11-18:
This seems to fail in STM32G0 as well (see #542).