renode/renode #419: Using device tree instead of the platform description language

mkschreder, 2022-12-16:
What about using the device tree format for platform description?

I believe it should be able to support all currently available features of the platform description syntax (and more).

It could potentially add much more standardization to renode platform definitions.

The way this can potentially be implemented is by calling libfdt directly from C#

Have there been discussions about adding this feature at all?

mithro, 2022-12-18:
Renode can generate a platform description from DeviceTree (https://opensource.antmicro.com/projects/dts2repl) - see https://zephyr-dashboard.renode.io/renodepedia/ for examples.

mkschreder, 2022-12-20:
Thanks @mithro. How do you render a trace like the one on this page: https://zephyr-dashboard.renode.io/renodepedia/boards/hexiwear_k64/?view=software&demo=Hello_World#colab

mithro, 2022-12-20:
@PiotrZierhoffer might know.

PiotrZierhoffer, 2022-12-20:
@mkschreder that's `cpu EnableProfiler`

Full documentation of the feature is here: https://renode.readthedocs.io/en/latest/advanced/execution-tracing.html#guest-application-profiling

In Renodepedia we use Speedscope traces.

closed 2023-01-03 by PiotrZierhoffer