renode/renode #421: How to expand "showAnalyzer uart" buffer?

gonbbb, 2022-12-22:
### Description

I confirmed with the `showAnalyzer uart` command that the firmware hangs after sending a 1023 bytes(in image) from uart. 

[image]

I looked for a variable somewhere that showAnalyzer allocates a buffer for, but couldn't find it.

This is the source code for the test, just outputting numbers. In reality, it outputs all sorts of characters.

### Expected behaviour

More characters are output.

### How to reproduce?

```
# resc
using sysbus
mach create
machine LoadPlatformDescription $ORIGIN/stm32f103.repl
showAnalyzer uart2
macro reset
"""
    sysbus LoadELF $ORIGIN/build/test_print.ino.elf
"""
runMacro $reset
start
```

```c
// src: test_print.ino
void setup() {
  Serial.begin(9600);
}

void loop() {
  print_data();
}

void print_data() {
  for (int i = 0; i < 10000; i++) {
    Serial.print(i);
    delay(100);
  }
}
```

compile method

```bash
arduino-cli compile ./test_print.ino -b=STMicroelectronics:stm32:GenF1
```

### Environment

- OS: macOS Big sur v11.4
- Renode 1.13.2(9ab53ff745cc4a3b908fe4bed3be6599631a88db)
- arduino-cli  Version: 0.29.0 Commit: 76251df9 

### Do you plan to address this issue and file a PR?

I just want to make adjustments in my personal environment, so either is fine.

Thank you.

gonbbb, 2022-12-23:
Sorry, It turns out that this occurred when `CreateFileBackend` was issued; without it, more strings were output. 

However, if it was too early, it HANGED. The later issue is related to issue #389

gonbbb, 2022-12-23:
`screen /path/to/dev_tty_file_by_CreateUartPtyTerminal` command resolved the hang. So I see that this issue addresses the same issue as #389 , and since it is a duplicate issue, I will close this issue.

The following image show that renode don't hang up after executing screen command

[image]

PiotrZierhoffer, 2023-01-03:
Hi @gonbbb 

Can you please clarify if your issue is related to PTY connection, or file backend, or both? Your last comment suggests PTY, but previous ones mention `CreateFileBackend`

gonbbb, 2023-01-04:
@PiotrZierhoffer 
This issue is related to PTY connection. 'file backend' is not relevant to this issue.