renode/renode #423: RISC-V CLIC support

madushan1000, 2023-01-01:
### Description

Support for RISC-V Core-Local Interrupt Controller (CLIC) (https://github.com/riscv/riscv-fast-interrupt/blob/master/clic.adoc). 

### Usage example

Looks like the Core-Local Interrupt Controller is still a draft proposal, but there are already a lot of MCUs using it (eg: sifive e2x core based chips like bl602)

### Additional information

Please add all information you feel to be relevant.

### Do you plan to address this issue and file a PR?

Maybe if it's not done already by somebody and I just can't find it.

madushan1000, 2023-01-04:
I'm trying to write CLIC interrupt controller for renode, I'm wondering if there is a "proper" way to set the CSR registers from a peripheral, I need to set RISC-V `MCAUSE` and bunch of other related registers. I was trying to use `machine.SystemBus.TryGetCurrentCPU(out var cpu)` and then `((ICPUWithRegisters)cpu).SetRegisterUnsafe(899, 0x80000009)`  on interrupt update. But looks like that doesn't work properly. Register value is unchanged. 899 is `MCAUSE` register.

madushan1000, 2023-01-04:
I looked into this a bit more, it looks like some changes to c modules are needed to add CLIC support.

PiotrZierhoffer, 2023-01-10:
At this moment we don't support any platforms with CLIC, that's why we don't have the model.

`MCAUSE` is a little bit specific - it is being updated in https://github.com/antmicro/tlib/blob/master/arch/riscv/helper.c#L378

I would assume this is a fairly generic method, but if there are changes to our translation libraries that CLIC would require to work - we'd be happy to discuss them. Can you share what kind of changes are you looking to add?

madushan1000, 2023-01-10:
I actually managed to get it mostly working without any changes to C code by taking over `mcause` register and installing custom handlers for `ecall` and `mret` in a custom cpu definition(https://github.com/madushan1000/bl602_renode/blob/main/RiscV32CLIC.cs). The mcu doesn't support other privilege modes. Most of what I need were already implemented for PicoRv32 interrupt support I think.

Now I'm hitting an issue with `wfi` instruction where all emulation halt but I'm not sure if this is an issue with the emulator or some other peripheral access I have to handle. I'm going to try installing a handler for `wfi` instruction too.

It could also be that I have implemented things wrong, I don't really understand how `TlibSetReturnRequest();`  and `TlibSetReturnOnException(1);` or any other calls into tlib work. 

Do you have chatroom where I can talk to developers about some of this?

madushan1000, 2023-01-11:
I figured out the wfi issue, I haven't wired the GPIOs correctly and the translation cpu wasn't receiving the timer interrupt.