renode/renode #425: Unhandled Exception when including cs files

xobs, 2023-01-16:
### Description

Something has changed on my system that prevents Renode from loading .cs files. Unfortunately, there isn't sufficient debug information to know exactly what the cause is, because the nightly builds do not have symbol information attached.

When starting a fresh Renode instance with `renode --console`, this happens:

```
(monitor) i @peripherals/BetrustedEcI2C.cs
17:34:31.0732 [INFO] Including script: E:\Code\Xous\Core\emulation\peripherals\BetrustedEcI2C.cs

Unhandled Exception: System.TypeInitializationException: The type initializer for 'Antmicro.Renode.Utilities.TemporaryFilesManager' threw an exception. ---> System.ComponentModel.Win32Exception: Access is denied
   at System.Diagnostics.ProcessManager.OpenProcess(Int32 processId, Int32 access, Boolean throwIfExited)
   at System.Diagnostics.Process.GetProcessHandle(Int32 access, Boolean throwIfExited)
   at System.Diagnostics.Process.get_HasExited()
   at Antmicro.Renode.Utilities.TemporaryFilesManager.IsProcessAlive(Int32 pid)
   at Antmicro.Renode.Utilities.TemporaryFilesManager.Cleanup()
   at Antmicro.Renode.Utilities.TemporaryFilesManager..ctor(String tempDirectory, String tempDirPrefix, Boolean cleanFiles)
   at Antmicro.Renode.Utilities.TemporaryFilesManager.Initialize(String tempDirectory, String tempDirPrefix, Boolean cleanFiles)
   at Antmicro.Renode.Utilities.TemporaryFilesManager..cctor()
   --- End of inner exception stack trace ---
   at Antmicro.Renode.Utilities.TemporaryFilesManager.get_Instance()
   at Antmicro.Renode.UI.CrashHandler.HandleCrash(Exception e)
```

### Expected behaviour

Renode should be able to include C# files.

### How to reproduce?

I'm unsure of how to reproduce. This might be an issue with my system, but I'm not sure how I would figure that out. Starting up Visual Studio just shows that the exception happened in "External Code":

[image]

### Environment

Please, provide the following information:

- OS: Windows 11 21H2
- Renode version: Renode, version 1.13.2.33150 (a4006479-202301121823)
- 
### Additional information

This may be due to something stale on my system. Perhaps a temporary file is missing? Or maybe there's a temporary directory that doesn't exist? I can't know because of missing symbols.

xobs, 2023-01-16:
Perhaps there was a user issue, but I went into `$env:TMP` and did a `takeown .exe /F *` followed by a `rm -recurse -force *`, after which things started working.

It may be that Renode tried to create a temporary directory as my user, after having previously created the directory as an Administrator. Even so, it would be nice for it to report *which* path it tried and failed to create, rather than simply exploding.

PiotrZierhoffer, 2023-01-24:
That's true. We've got similar reports and it's quite unclear what happens. 

We're also planning to make the temp dir configurable for users, as sometimes it's difficult to use the default (because of various mount options as in #415 )