renode/renode #427: The top line of text blinks when two console analyzers are open

xobs, 2023-01-27:
Whenever two console analyzers are open, the top line of text blinks. The two console windows alternate their blinking.

It's always the top visible line that blinks.

## Steps to reproduce

1. Create two serial ports
2. Call `showAnalyser` on both serial ports

AntonKrug, 2023-01-27:
Could you be more specific? Your OS, versions of things? I'm just a user, I recall having multiple ports open often and can't recall seeing this issue.

xobs, 2023-01-27:
Sure, apologies for the vague bug report.

This is Windows 11 21H2, using Renode 1.13.2.33150 (a4006479-202301121823)

Can you try resizing one of the windows after calling `showAnalyzer`?

xobs, 2023-03-12:
Apologies, it seems as though this is not fixed by 28aa9fd -- I see text blinking again.

xobs, 2023-04-20:
Apparently an issue that's exacerbated by adjusting the width of the window. It happens whenever there is more than one window open.

Here is a video with four windows. It seems as though one window will always have the top line missing, but it's random as to which one. The line shifts every ~500ms it seems.

https://user-images.githubusercontent.com/238325/233269963-8f69d898-e1eb-4fe5-b7c7-20f9dab8dfae.mp4

xobs, 2023-12-28:
I've installed the latest nightly, and the windowing system seems different. It also looks like this issue is fixed now!

closed 2023-12-28 by xobs