renode/renode #428: Discussion email list or forum

jasonbyun8, 2023-01-31:
I have been enjoying using renode so far, but if I come across to any questions, I cannot find any good place to search or ask except the doc from renode website.

QEMU comes with discussion mail list --  https://www.qemu.org/support/ -- which really helps to learn material quickly and enables user to help each other. 

It would be really great if renode also comes with such features; email list, forum, discord, telegram, etc.

sulemankm, 2024-02-09:
Yeah me too!  Renode seems to be a very good piece of software.  However, a public discussion/chat forum is direly needed and would be great to have!

otavio, 2024-02-09:
Agreed. This would help in the tool adoption.