renode/renode #429: Unable to copy text

xobs, 2023-02-02:
### Description

Sometime recently, the ability to copy text out of the serial analyzer has stopped working. It used to be that right-clicking would bring up a box from which you could select "Copy", "Paste", or "Do not append...". However, now the box disappears immediately, making it impossible to select anything.

### Expected behaviour

It should be possible to copy text from the analyzer, either by a key combination or by right-clicking.

### How to reproduce?

Attempt to right-click to copy data in an analyzer

### Environment

Please, provide the following information:

- OS: Windows 11
- Renode version (with commit SHA): Renode, version 1.13.2.33150 (a4006479-202301121823)

tfirchau, 2023-02-02:
Hi! I have observed the same problem on 

- OS: Windows 10
- Renode version:  Renode, version 1.13.2.5472 (9ab53ff7-202210030301)

It still works for me in Renode, version 1.13.1.21690 (a133ada0-202207251201)

mateusz-holenko, 2023-03-09:
Thanks @xobs for reporting the issue.
Could you verify if 28aa9fd2a169a5fb35621618d7d7e0372cc81f16 resolves the problem for you?

xobs, 2023-03-09:
28aa9fd does not solve this issue. However, it does solve #427

PiotrZierhoffer, 2023-03-21:
This is indeed quite strange. We were able to reproduce this locally, and after the fix the issue is gone on our end - apparently it was a race. We'l look further into reproducing it with the latest Renode, but it might not be trivial.

xobs, 2023-04-20:
This appears to be fixed now.

closed 2023-04-20 by xobs