renode/renode #431: Support for 64-bit ARM

podhrmic, 2023-02-02:
### Description

Are you planning to add support for 64-bit ARM CPUs? Specifically ARMv8A architecture with ARM HYP virtualization, and either Cortex-A53 or Cortex-A72? In other words, will it be possible to emulate Raspberry Pi 3/4 in Renode?

And if it is on the roadmap, what is the expected timeline?

I really like Renode, and I am using it for some seL4 testing, but the majority of seL4 [supported platforms](https://docs.sel4.systems/Hardware/) are 64-bit ARM.

PiotrZierhoffer, 2023-02-03:
Hi @podhrmic 

We're working on ARMv8A, and soon we should have an initial version released.

We're also looking into seL4 in this context, as it's a good testing layout. 

I think we should have more updates on this topic very soon.

PiotrZierhoffer, 2023-04-04:
Hi @podhrmic,

Pleased to announce we have just released preliminary support for ARMv8-A. You will find a Zephyr demo targeting generic Cortex-A53 here https://github.com/renode/renode/blob/master/scripts/single-node/cortex-a53.resc

This is still work in progress, we are currently working on GIC and interrupt support in general, but you should expect more capabilities soon.

As for specific platforms, this is driven by customer demand and our own needs. But since this issue concerns ARM64 in general, I assume we can close it as done.

closed 2023-04-04 by PiotrZierhoffer