renode/renode #435: Execution Tracer couldn't disassemble opcode

BobbyCounts, 2023-02-16:
### Description

I'm mainly interested in renode for tracing CPU execution through various MCU workloads and using the collected trace for modeling performance. I'm using ```scripts/single-node/max32652-evkit.resc``` to load a basic hello_world example with execution tracing:
```
cpu EnableExecutionTracing @./trace.txt PC
```
Whenever I start the simulation I get a storm of error messages like below:
```
13:53:35.7384 [WARNING] cpu: ExecutionTracer: couldn't disassemble opcode at PC 0x10000926 (787282)
```
Also, whenever I try to quit, renode stops responding to keyboard input and takes up to several minutes to exit due to the amount of warning logs generated. If I disable the warning log generation in the ExecutionTracing source this issue goes away.

### Expected behaviour
I expect the tracer opcode decoding to work since the simulation seems to work correctly without tracing and the simulation must be aware of the opcode to emulate the processor execution.

### How to reproduce?
Run
```
./renode scripts/single-node/max32652-evkit.resc
```
```
(machine-0) cpu EnableExecutionTracing @./trace.txt PC
(machine-0) s
```

### Environment

Please, provide the following information:

Ubuntu 22.04.1 LTS x86_64
Renode 1.13.2.24742 (9ab53ff7-202302161344)

BobbyCounts, 2023-02-16:
Also, within TraceFormat.cs, it seems that no matter the format selected the program always tries to decode and disassemble the instruction. Perhaps if format == PC, the PC can be directly written to the file without the decode stage to improve performance.

JakubJatczak, 2023-02-24:
Hi @BobbyCounts, 
Thanks for submitting this, execution tracing for the cortex-m4f cores was indeed broken. I pushed a fix at 2b8fd87 and current master should now work fine.
Regarding the unresponsiveness right after quitting, this is an effect of waiting for the tracing thread to finish processing execution data, and it only took so long because of the issue you found. We also print log informing user that Renode will quit as soon as the tracing is done, but in your case it was covered with all the warnings.
I'm closing the issue, thanks again for spotting this

closed 2023-02-24 by JakubJatczak