renode/renode #440: Zynq7000: Could not attach PHY

podhrmic, 2023-03-08:
### Description

The [Zynq7000 example](https://github.com/renode/renode/blob/master/scripts/single-node/zynq-7000.resc) doesn't seem to have working networking. 

### Expected behaviour

`ip link set eth0 up` should bring up `eth0` device, and network comms with host should work.

### How to reproduce?

* start renode
* load `scripts/single-node/zynq-7000.resc`
* start simulation
* create and initialize `tap0` on your host
* in renode, attach `tap0` with ` emulation CreateTap "tap0" "tap"`
* in emulated linux, run:
```
# ip addr add 172.16.0.3/24 dev eth0
# ip link set eth0 up
```
* You will see:
```
# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop qlen 1000
    link/ether 00:00:00:00:00:02 brd ff:ff:ff:ff:ff:ff
    inet 172.16.0.3/24 scope global eth0
       valid_lft forever preferred_lft forever
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
# ip link set eth0 up
macb e000b000.ethernet eth0: Could not attach PHY (-19)
ip: SIOCSIFFLAGS: No such device
```
* In renode stdout:
```
11:29:36.6855 [INFO] Opened interface tap0.
11:31:30.8639 [WARNING] pl310: Unhandled read from offset 0xF40. (6)
11:31:30.8641 [WARNING] gem0: Unhandled write to offset 0x0. Unhandled bits: [5] when writing value 0x20. Tags: clear_all_stats_regs (0x1).
11:31:30.8641 [WARNING] gem0: Unhandled write to offset 0x14. Unhandled bits: [1-4, 6-31] when writing value 0xFFFFFFFF. Tags: collision_occurred (0x1), retry_limit_exceeded (0x1), transmit_go (0x1), amba_error (0x1), transmit_under_run (0x1), late_collision_occurred (0x1), resp_not_ok (0x1), RESERVED (0x7FFFFF).
11:31:30.8642 [WARNING] gem0: Unhandled write to offset 0x20. Unhandled bits: [2-31] when writing value 0xFFFFFFFF. Tags: receive_overrun (0x1), resp_not_ok (0x1), RESERVED (0xFFFFFFF).
11:31:30.8647 [WARNING] gem0: Unhandled write to offset 0x2C. Unhandled bits: [4-6, 8-31] when writing value 0xFFFFFFFF.
11:31:30.8652 [WARNING] gem0: Unhandled write to offset 0x24. Unhandled bits: [4-6, 8-31] when writing value 0xFFFFFFFF.
11:31:30.8653 [WARNING] gem0: Unhandled write to offset 0xD4, value 0x0.
11:31:30.8653 [WARNING] gem0: Unhandled write to offset 0xD8, value 0x0.
11:31:30.8654 [WARNING] gem0: Unhandled write to offset 0x90, value 0x0.
11:31:30.8654 [WARNING] gem0: Unhandled write to offset 0x94, value 0x0.
11:31:30.8654 [WARNING] gem0: Unhandled write to offset 0x98, value 0x0.
11:31:30.8654 [WARNING] gem0: Unhandled write to offset 0x9C, value 0x0.
11:31:30.8654 [WARNING] gem0: Unhandled write to offset 0xA0, value 0x0.
11:31:30.8655 [WARNING] gem0: Unhandled write to offset 0xA4, value 0x0.
11:31:30.8655 [WARNING] gem0: Unhandled write to offset 0x4. Unhandled bits: [8, 18-19, 24] when writing value 0x1068100. Tags: receive_1536_byte_frames (0x1), mdc_clock_division (0x3), receive_checksum_offload_enable (0x1).
11:31:30.8658 [WARNING] gem0: Unhandled write to offset 0x10. Unhandled bits: [2, 4, 6-7, 17, 19-20] when writing value 0x180F50. Tags: amba_burst_length (0x4), endian_swap_management (0x1), endian_swap_packet (0x1), rx_buf_size (0x1A).
11:31:30.8681 [WARNING] gem0: Unhandled write to offset 0x0. Unhandled bits: [5] when writing value 0x20. Tags: clear_all_stats_regs (0x1).
11:31:30.8681 [WARNING] gem0: Unhandled write to offset 0x14. Unhandled bits: [1-4, 6-31] when writing value 0xFFFFFFFF. Tags: collision_occurred (0x1), retry_limit_exceeded (0x1), transmit_go (0x1), amba_error (0x1), transmit_under_run (0x1), late_collision_occurred (0x1), resp_not_ok (0x1), RESERVED (0x7FFFFF).
11:31:30.8681 [WARNING] gem0: Unhandled write to offset 0x20. Unhandled bits: [2-31] when writing value 0xFFFFFFFF. Tags: receive_overrun (0x1), resp_not_ok (0x1), RESERVED (0xFFFFFFF).
11:31:30.8682 [WARNING] gem0: Unhandled write to offset 0x2C. Unhandled bits: [4-6, 8-31] when writing value 0xFFFFFFFF.
11:31:30.8682 [WARNING] gem0: Unhandled write to offset 0x24. Unhandled bits: [4-6, 8-31] when writing value 0xFFFFFFFF.

```
* Could be related to #439 as the same `CadenceGEM` device is used for phy
### Environment

Please, provide the following information:

- OS Ubuntu 20.04
- Renode version: 1.13.3 a72a1fa1eff75b8ce9f2a7067ef12e8b2f402281

podhrmic, 2023-03-08:
Note that I can successfully communicate over network with [HiFive Unleashed](https://github.com/renode/renode/blob/master/scripts/single-node/hifive_unleashed.resc). HiFive has a different `phy` peripheral, maybe that is relevant?

sand7000, 2023-03-09:
Nevermind, I missed the part where I have to connect the device to the switch after the machine has started.

PiotrZierhoffer, 2023-04-04:
@podhrmic we have updated the platform description and it works for us internally. We will release this soon, and let you know

vannaka, 2023-11-09:
@PiotrZierhoffer Any update on the fix you mentioned? I looked through the [history](https://github.com/renode/renode/commits/master/platforms/cpus/zynq-7000.repl) of zynq-7000.repl and see no relevant changes.

closed 2024-01-10 by MateuszKarlic