renode/renode #441: riscv: `ebreak` instruction incorrectly issues `ecall` exception

xobs, 2023-03-14:
Renode is generating `ecall` exceptions when it encounteres an `ebreak` instruction. This makes it impossible to tell whether a trap was made due to a syscall or a breakpoint, making it potentially more expensive inside of a kernel.

### Description

According to the [Risc-V privileged spec](https://riscv.org/wp-content/uploads/2017/05/riscv-privileged-v1.10.pdf) section 3.2.1:

[image]

However, Renode appears to be generating an `ecall` exception regardless of whether the opcode is `ecall` or `ebreak`.

### Expected behaviour

`ebreak` should generate a `Breakpoint` exception, and `ecall` should generate a call exception of the appropriate level.

### How to reproduce?

I have created [a gist](https://gist.github.com/xobs/218adafdd1b7b3670561e845e10ee4a3) that contains code that demonstrates the problem. The output is reproduced here:

[image]

When run under TinyEmu, the following output is produced, which I believe is correct:

```
!starting test...
ecall instruction from machine mode
ebreak instruction
ecall instruction from machine mode
ebreak instruction
finished test

Power off.
```

Instructions for compiling the example program are included at the top. The machine I'm running tests in is:

```
mach create
using sysbus
machine LoadPlatformDescriptionFromString "cpu: CPU.VexRiscv @ sysbus { cpuType: \"rv32gc\"; timeProvider: empty }"
machine LoadPlatformDescriptionFromString "mem: Memory.MappedMemory @ sysbus 0x80000000 { size: 0x1000 }"
machine LoadPlatformDescriptionFromString "uart: UART.LiteX_UART @ sysbus 0x40008000"
sysbus LoadELF @my.elf
showAnalyzer uart
start
```

### Environment

Please, provide the following information:

- OS: Windows 11
- Renode version (with commit SHA): Renode, version 1.13.3.6128 (e9611fb4-202303090322)

### Additional information

It is possible to read the offending instruction and determine whether it was a `ebreak` or `ecall`, however that requires looking into the memory of the generating program which may not be possible due to memory limitations, or would incur a large performance penalty.

mateusz-holenko, 2023-03-30:
Thanks for reporting the issue, @xobs.

`c.ebreak` exception generation is now fixed in https://github.com/renode/renode/commit/b7431ca4c791c02c1d792084469e28f2793d69f5.
Could you verify if it works for you?

xobs, 2023-04-05:
Thanks, this does fix the issue.

closed 2023-04-05 by xobs