renode/renode #445: Counting CPU cycles isn't accurate

frejohg, 2023-03-18:
HiT

I'm trying to measure the time it takes to run a function.
I'm comparing hardware and Renode.
For my arduino nano 33 sense ble I can use the built in cycle counter and output this on serial as following:

DWT->CYCCNT = 0;
start = DWT->CYCCNT;
myFunction();
stop_ = DWT->CYCCNT;
elapsed = stop_-start;
Serial.println(elapsed);

I get around 25000 cycles.

If I use the binary in Renode

machine LoadPlatformDescription @/home/fred/appimages/renode_1.13.2_portable/platforms/boards/arduino_nano_33_ble.repl
sysbus LoadELF @/home/fred/Arduino/hello_world_cyccnt2/build/arduino.mbed_nano.nano33ble/hello_world_cyccnt2.ino.elf
machine StartGdbServer 3333 true
sysbus.cpu EnableProfiler CollapsedStack @/home/fred/trace1
machine EnableProfiler @/home/fred/profiler1

I get around 12000 cycles when checking in speedscope.

I had a thought that one of them might be considering both cores of the arm, but it does not seem that Renode has exactly twice the amount of cycles. I've tried this with several different examples of functions.

Enviroment:
arch x86_64
xfce
Renode, version 1.13.2.18007 (9ab53ff7-202210031000)
arduino-ide_2.0.4_Linux_64bit

PiotrZierhoffer, 2023-03-21:
Renode is a functional simulator, and while there are ways to make it more accurate in terms of time measurement, it's not aiming to be 1:1 with hardware.

You have two things you can check: 

- `cpu PerformanceInMips`  value serves as a way to calculate how many instructions we execute in a unit of time. You can easily change it after loading the repl file.
- clock frequencies - they are defined inside repl, and there may be something off. DWT, specifically, has its own frequency set there, it might be an invalid value.

Please verify it with the docs and let us know if you find better settings for you

frejohg, 2023-03-22:
Hi

Ok I tried the PerformanceInMips and found that it is not linear, meaning that it does not scale accurately, I could find a setting that produced the same results as tests on a physical device, but this setting did not produce accurate results between renode and  another calculation on the same device.

Could you specify which of the settings in the nRF52840.repl file that are the clock cycles? That would be helpful thank you.

PiotrZierhoffer, 2023-04-04:
@frejohg I assume the most reasonable option for you would be to add a line here: https://github.com/renode/renode/blob/master/platforms/cpus/nrf52840.repl#L2

```
nvic: IRQControllers.NVIC @ sysbus 0xE000E000
    systickFrequency: 10000000 // your frequency here
    -> cpu@0
```