renode/renode #450: Question about the ability of renode

fanyx-ok, 2023-04-10:
### Description
I have a custom device using stm32 chip and some other peripheral control chip, including UART, LED, BT, BLE, Charger and some other functionalities.  is renode able to simulate this device if I code some rule for my device? 

### Usage example
to increase develop efficiency.

Thanks.

PiotrZierhoffer, 2023-05-16:
Hi @fanyx-ok 

Sure, Renode can simulate both the SoC and peripherals you attach to it. You can also have multi-node setups, connected via various means, if your control chip is an independent being.

Take a look at https://renode.readthedocs.io/en/latest/advanced/writing-peripherals.html and the models we have in https://github.com/renode/renode-infrastructure/tree/master/src/Emulator/Peripherals/Peripherals for inspiration.

Antmicro also provides engineering services if you'd require help in writing models for your devices - feel free to reach out via email.

fanyx-ok, 2023-05-17:
Thanks, I will have a try.

closed 2023-05-17 by fanyx-ok